OceanDataTools/openrvdas #31: Make code compatible with Python 3.2.3 for Raspbian

davidpablocohn, 2017-11-30:
Because, you know, DASBerry Pi...

webbpinner, 2017-11-30:
What is lost in terms of performance and functionality by downgrading the version of python from 3.5.x to 3.2.3?

Is there the RPi community going to be updating python anytime in the near future?

davidpablocohn, 2017-11-30:
It looks like most of the changes are very minor - things like str.split() not taking keyword arguments, and certain exceptions not existing. Also in parallel started installing python 3.6 on my (ancient) Pi to see how hard that was. Figuring out which is the path of least resistance is probably a good modular project. But as discussed on separate thread, the benefits of 3.5+ probably outweigh the (minor) pain of requiring a newer version of Python, so maybe this is moot.

webbpinner, 2017-12-03:
I'd vote for requiring Pi users to upgrade vs foregoing all the features and performance advantages found in python version 3.5+.

davidpablocohn, 2017-12-03:
Agreed!

closed 2017-12-03 by davidpablocohn