OceanDataTools/openrvdas #35: JSON config files should use Javascript-style // comment prefixes

davidpablocohn, 2017-12-03:
JSON itself doesn't support comments, but we're using JSON for config files, and config files really need comments.

The logger.utils.read_json scripts strip out comments beginning with "#", in Python style, but JSON is *Javascript* Object Notation, so comments should really be Javascript format.

closed 2017-12-03 by davidpablocohn

webbpinner, 2017-12-04:
I know this issue is marked as closed but just for thought... was YAML ever considered as a format for config files?

davidpablocohn, 2017-12-04:
Hadn't considered YAML. Did discuss XML and discarded it. But read_json() is only called in five files, and it should be pretty easy to tweak things to allow a read_yaml() function to be swapped in (hurrah for functions being assignable!). Pretty much anything that can convert text -> Python dict should work.

commit d7a9056f (2017-12-03): Convert comment character in JSON files from # to //, in keeping with Javascript origins of JSON. Resolves #35.