OceanDataTools/openrvdas #36: Open discussion on requirements for a Graphical User-Interface

webbpinner, 2017-12-06:
I wanted to start the discussion on drafting requirements for a TBD graphical user-interface (GUI) for OpenRVDAS.

In the current state of OpenRVDAS (December 6, 2017) there is no GUI.  From my own experiences developing the OpenVDM GUI I understand the importance of drafting requirements before committing to a development platform/architecture and commence writing code.  The purpose for opening this issue is to capture the community's requirements from a GUI for an underway data logger.

Inevitably the final design requirements may not meet all the needs and wishes of every user but hopefully by the end of this discussion there will be a well defined path towards a majority agreed upon solution.

davidpablocohn, 2017-12-07:
Thanks for starting this thread up! Here are some initial thoughts:

The 1.0 desiderata we worked toward on the USAP system included the ability to:
1. configure loggers (define a cruise, and the behavior of each logger for different phases of that cruise).
2. start/stop loggers
3. monitor "health" of loggers via alarms/screen alerts/etc
4. configure and launch data quality monitors
5. configure and launch a variety of data displays

Non-GUI management functionality included
1. configure processes for generating derived-data (e.g. true winds) from loggers
2. end of cruise packaging and distribution of accumulated data

webbpinner, 2017-12-07:
Not really a UI-related comment...
I agree with the need/desire to have an organizational unit (Cruise) for data loggers.  However I do suggest using a more general term than "cruise".  This would allow groups needing a data logger for things other than cruises.  I'm thinking ROVs, AUVs, HOVs, etc, where the operator may wish to configure loggers within a container that is not "a cruise".

I recently ran into this situation with The R/V Falkor and their ROV SuBastian.  They had to setup a second DAS specific to the ROV that organizes the data from the ROV by dive.

webbpinner, 2017-12-07:
Related to #2
A feature request I hear from mostly SCS users is the ability to restart individual loggers.  The need for this functionality in a UI is coupled to one of the core "features" of SCS... if you make a change to any of the loggers, all the loggers must be restarted.

...a little of topic.
I agree that the ability to start/stop individual loggers would be a great feature to have in a UI, but I also understand why SCS's UI doesn't.  I'm not 100% confident in exactly how SCS works under the hood but I believe SCS stored the most recent values from all loggers in a page-file or some other shared data structure.  If the configuration for a logger is changed, it potentially could change the structure of the shared data structure which in-turn could break an inter-dependencies and ultimately compromise/crash the DAS.

I don't know if you experienced this first-hand on the NBP.  I also don't know if implementing this feature in the UI would introduce architectural changes to the loggers.

webbpinner, 2017-12-07:
Related to #3, #4, and #5... sorta
This is going to be the most requested feature in a UI... and the one everyone will want to customize for their particular platform and operating procedures.

I think the initial offerings should include a general numerical display with visual indicators for:
 - no new data received for a defined amount of time
 - malformed incoming data
 - data out of acceptable range

This is what the WHOI ships currently have on monitors around the ship... I'm not sure if it's implemented as a web-page or app but I think their using something like a RPi3 for the client hardware.

I think notifications will also be popular... via email and/or something else.

From there the sky's the limit for how for you want to take things, i.e. graphs, widgets, gauges, etc.

However before I went too far down that road I'd recommend first implementing an API that allows users to subscribe to data feeds.  I see this as something different than network writers.  I'm thinking RESTful API, websockets, etc.  While this isn't a GUI per say it is a UI that enables users to "monitor" what's happening.  A lot of the ships I go on have invested a substantial amount of time implementing some kind of vessel-wide data display system.  Enabling the techs that support (and probably developed) these vessel specific systems to adapt their existing systems to simply pull from a new data source will most likely be more attractive than providing them with an entirely new system that they must adapt to.

In my experience techs are hackers at heart... don't give them a turn-key solution, give'em something to play with that allows them to develop something they can take pride in.

davidpablocohn, 2017-12-07:
Excellent point! So I need to find a better term than "cruise" - "deployment"? "run"?

The USAP code only ever stopped/restarted whichever logger needed changing. The model was that the database maintained a history of "observed state" and "desired state" for each logger (see slide 16 of https://www.unols.org/sites/default/files/201611rvt_BreakOut_DAS_USAP.pdf). The job of the logger manager was to poll and make sure each logger was in its desired state, and if not, do something about it. All the web interface did was insert new "desired states" when the user hit a button. The logic flow for this was all surprisingly clean, and I'm hoping to make use of some "lessons learned" there. I will plan to draw up a diagram.

closed 2024-08-23 by davidpablocohn