OceanDataTools/openrvdas #42: Initial Django-based GUI implementation

davidpablocohn, 2017-12-24:
Related to #36 - roadmap of initial implementation.

0.1. Load cruise config from file, display modes, and select modes, save back to file. 

0.2. Display will show all defined loggers and each config that exists for a logger. Allow selecting individual configs, then saving aggregated configs as a new mode.

0.3. Actually run/monitor the loggers as specified by the current config.

0.35. Save a history of current config to logfile

0.4. Allow editing of config vars (cruise id, logfile location, etc)

webbpinner, 2017-12-24:
I completely agree that staying with python for the UI makes sense.  Is there an advantage/need for a fully fledged MVC framework like Django vs going with a micro framework like Flask?

davidpablocohn, 2017-12-24:
Good question. In USAP, we picked Django arbitrarily, and were psyched to find the OSU folks using it, too. Don't really *need* Django's ORM and more esoteric features, but maximizing compatibility w/OSU is a priority for me. Of course, nothing to say that folks can't try GUIs in different languages - my roadmap calls for keeping the openrvdas/manager interactions API-driven.

Maybe worth throwing a query out to the mailing list to get a show of hands?

commit 968743f8 (2018-06-27): Final tweaks to get build script and services running on CentOS 7: clean up Javascript, fix broken unittest port and add missing specification that openrvdas service should use the Django database. Resolves #67 and resolves #42.
closed 2018-06-27 by davidpablocohn