OceanDataTools/openrvdas #51: GUI Implementation questions

webbpinner, 2018-02-05:
These questions were raised by Pablo regarding next steps in the development of the OpenRVDAS GUI: 
- Should the gui use websockets or Django channels for server-client communications

- Should the LoggerServer run as a stand-alone process, as a service, or stopped/started via a button on the GUI console?

webbpinner, 2018-02-05:
My thoughts...
Regarding the websockets/Django channels question...
I guess I don't have a preference so long as the final solution can accommodate connections from non-Django sources.  I don't know enough about Django channels to make an educated on this.  From my experience with websockets I know it's easy enough to connect via javascript running in a webpage or with python script running in a terminal.

Regarding the how to run the logger and gui....
For myself the most important thing is to have a solution that provide the needed level of control.  That control being:
 - Creating a new logger config
 - Making changes to an existing logger config
 - Enabling/Disabling logging

I also see the benefit in there being separation between the gui and logger.  By that I mean I think a user should be not be required to run any gui if the user wanted to save on memory, processing overhead, etc.

A second advantage in logger/gui separation is that it would allow users to develop their own GUI using their preferred architecture.

Now the caveat with both of my preferences is that in order for the second perference to work there would need to be some sort of API to allow a stand-alone gui or a 3rd-party developed gui... this of course means the core of the logger would need to include a minimal webserver service for serving the API call responses.  So I guess my vote on this matter would be as follows:

 - have the logger include a minimal webserver for the purpose of serving an API, if desired.
 - have a variable in a config file that controls whether or not enable that API.

This is similar to how Supervisor works... supervisor provides a web-interface that can be disabled via it's config file.  The config file is also used to set the port number to host the web-interface on.  This is very useful when needed to manage multiple services and I could see it being needed for OpenRVDAS should the user want to run multiple instances on the same machine.

davidpablocohn, 2018-02-10:
Thanks, Webb!

My understanding is that "channels" are just a Django implementation of websockets, so widgets, etc. trying to connect shouldn't see any difference. The main difference would be if one wanted to entirely avoid using the Django server as a backend, as per option (2) below.

In terms of logger/gui separation: there's already a separation with the command line version of the controller (logger/utils/run_loggers.py), and the Django version (gui/django_run_loggers.py). But the two are parallel implementations, rather than the latter built on the former.

The command line version reads in a configuration file and runs the loggers specified in it. The gui version loads a configuration file into Django's model/database; django_run_loggers.py examines the database for changes (desired_state vs. observed_state) and updates what's running based on that. The actual gui part serves state from the database and takes HTTP POST requests that modify the database.

If you want to maintain logger state persistently (not in memory as the command line run_loggers.py does), right now the Django DB is the only way to do that. You don't have to use the Django webserver to interact with it, though. A few options:

1) still use the Django server to field POST requests and make the DB modifications
2) enhance django_run_loggers.py to take all the database-modifying commands itself. The previous incarnation of the websocket-based code did this, but I never did figure out how to do user authentication on websockets.

Thoughts? How hard should I try to make this branch independent of the Django backend?

webbpinner, 2018-02-11:
If Django "channels" are directly analogous to websocket pubs/subs then I think the current approach should work fine.

While I personally would have preferred a web framework more lightweight than Django, I'm really not a fan of trying to maintain 2 versions of OpenRVDAS this early in the project's lifespan.  At a later date this group may want/need to revisit this decision but for now I'd vote to move forward with what's in place.

My real concern is the need for `run_loggers.py` and `django_run_loggers.py`.  In an ideal world there would only be one entry point and the command-line arguments completely define the process's behavior ( i.e. run_logger.py -f <command file> [-w <webserver url>] ).  The way you described it sounds like a config file is being loaded in both scenarios... so is the program call I described possible?

davidpablocohn, 2018-02-12:
> [-w ] ). The way you described it sounds like a config file is being loaded

Yup - I think having a single core executable with/without gui makes sense.
The challenge is how to persistently store state without a database. I'm
going to stare and squint at the way I've got things implemented now to see
if I can find a clean way to do it. In the meantime, I'll try decoupling
the execution and DB code as much as possible in django_run_loggers.py to
facilitate that.

webbpinner, 2018-02-14:
is leveraging a db such a bad thing?

I'm thinking more of a minimalist DB like SQLite which I believe is just a file with some sort of access wrapper to prevent simultaneous write.  Give this a read: <https://www.sqlite.org//whentouse.html>.

davidpablocohn, 2018-02-14:
Interesting idea. SQLite is the default DB for Django and works. What would
you think of this for portability: relying on Django-implemented models for
persistent state management, but being agnostic about actual GUI? As you'd
earlier suggested, the executable could have a flag saying "I don't care
about persistent state, just run stuff" (the way
logger/utils/run_loggers.py currently does). If specified, the code
wouldn't try to load/use Django.

webbpinner, 2018-02-14:
What I was thinking was that the state information be stored in something like SQLite but not tied to a 3rd-party schema.  I don't know enough about Django.  Can Django pull data from a DB other than it's own? i.e. can there be an OpenRVDAS schema that is used by the CLI and an optional GUI?

davidpablocohn, 2018-02-14:
Ah, yeah - got it. Interesting thought.

You can define pretty much any backend DB you want for Django. And inside
the Python code that you write for Django you could certainly make manual
calls to a DB that wasn't associated with Django. But a lot of the
ease-of-use of Django comes from that seamless DB integration. I'm in
Corvallis for a few days to work with the OSU folks - lemme bounce this
idea around with them to see if I can convince myself that it's doable
without any unnatural contortions?

webbpinner, 2018-02-14:
Which is easier in the long-run... 
 - Supporting a Django-specific and CLI-specific version of OpenRVDAS
 - Supporting one version of OpenRVDAS that is interface agnostic but requiring all user-interfaces (CLI, Django or otherwise) to leverage/comply with an OpenRVDAS-specific schema?

Again, I don't know Django and to date haven't had time to pour through the repo to understand the repercussions of what I'm suggesting.

davidpablocohn, 2018-02-14:
Okay, I think, for the next few weeks, I'm going to risk taking a step down
the slippery slope of merging the current Django interface into the main
branch. I'm going to try to compartmentalize the Django and DB components
as much as possible with an eye toward being able to swap them out for a DB
that is not tied to a GUI.

Yes, there's a danger that I'll find myself in a corner I can't climb out
of and I say Dang, I think we're all tied to Django (Dango?). But I think
that risk is low enough that I'm willing to risk it to make I'm wanting to
make forward progress on closing the GUI loop.

davidpablocohn, 2018-11-13:
I think we've solved this one with the API separation allowing HAPI/Django/in-memory implementations.

closed 2018-11-13 by davidpablocohn