OceanDataTools/openrvdas #62: Alter run_loggers/LoggerServer to comply with API

davidpablocohn, 2018-04-14:
Set up core class to have methods that implement the following calls:
(NOTE: names of calls will probably change, but that's okay - what we're after are methods that implement the functionality)

- GET openrvdas_api/api/v_2a/configs
  - get current configs LoggerServer is running
- POST openrvdas_api/api/v_2a/configs
  - set configs LoggerServer should be running
- GET openrvdas_api/api/v_2a/configs/{id}
  - get current config LoggerServer is running for logger {id}
- POST openrvdas_api/api/v_2a/configs/{id}
  - set current config LoggerServer should run for logger {id}
- GET openrvdas_api/api/v_2a/configs/{id}/start
  - enable current config LoggerServer is running for logger {id}
- GET openrvdas_api/api/v_2a/configs/{id}/stop
  - disable current config LoggerServer is running for logger {id}
- GET openrvdas_api/api/v_2a/configs/{id}/restart
  - restart current config LoggerServer is running for logger {id}
- GET openrvdas_api/api/v_2a/status
  - get status, including error messages, for all current configs
- GET openrvdas_api/api/v_2a/status/{id}
  - get status, including error messages, for logger {id}

webbpinner, 2018-05-15:
I'll try to provide the equivalent calls in the current API

**GET openrvdas_api/api/v_2a/configs - get current configs LoggerServer is running**
_GET http://<API Server:Port\>/openrvdas_api/api/v1/loggers/active?loggerServer=<loggerServerID\>_

**POST openrvdas_api/api/v_2a/configs - set configs LoggerServer should be running**
The logger configs currently contian the loggerServerID of the loggerServer that should run them. The loggerServerID is currently stored in the logger configs because a logger config can't safely be run by multiple loggerServers.  To update the loggerServerID within a logger config:
_POST http://<API Server:Port\>/openrvdas_api/api/v1/loggers/<loggerConfigID\>_
This call includes a post payload of:
_{
  "loggerServer":"<loggerServerID\>"
}_

**GET openrvdas_api/api/v_2a/configs/{id} - get current config LoggerServer is running for logger {id}**
I don't understand this call.  Can you please explain this call a little more.

**POST openrvdas_api/api/v_2a/configs/{id} - set current config LoggerServer should run for logger {id}**
I don't understand this call.  Can you please explain this call a little more.

**GET openrvdas_api/api/v_2a/configs/{id}/start - enable current config LoggerServer is running for logger {id}**
_GET http://<API Server:Port\>/openrvdas_api/api/v1/loggers/<loggerConfigID\>/start_

**GET openrvdas_api/api/v_2a/configs/{id}/stop - disable current config LoggerServer is running for logger {id}**
_GET http://<API Server:Port\>/openrvdas_api/api/v1/loggers/<loggerConfigID\>/stop_

**GET openrvdas_api/api/v_2a/configs/{id}/restart - restart current config LoggerServer is running for logger {id}**
_GET http://<API Server:Port\>/openrvdas_api/api/v1/loggers/<loggerConfigID\>/restart_

**GET openrvdas_api/api/v_2a/status - get status, including error messages, for all current configs**
Status (.i.e running/stopped ) is available via the logger configs.
This returns all the logger configs: _GET http://<API Server:Port\>/openrvdas_api/api/v1/loggers_
This returns all the logger configs for the active mode: _GET http://<API Server:Port\>/openrvdas_api/api/v1/loggers/active_

**GET openrvdas_api/api/v_2a/status/{id} - get status, including error messages, for logger {id}**
Status (.i.e running/stopped ) is available via the logger configs.
_GET http://<API Server:Port\>/openrvdas_api/api/v1/loggers/<loggerConfigID\>_

davidpablocohn, 2018-05-17:
Thanks! But please hold off just a little. I'm still tweaking the "**status**" message format, as I'm realizing the notion of "last N status messages" isn't as well-defined (or useful) as I'd thought. Mostly because not all loggers are necessarily going to get checked at the same time, so assembling "last N for each logger" is challenging. Currently thinking of replacing with "status" returning most recent status for each logger and "**status_since [timestamp=0]**" returning all status updates since specified timestamp (with timestamp=0 as default). This will allow user to iteratively get all status updates if they want by specifying the latest status timestamp retrieved on a previous call and using it for the next.

davidpablocohn, 2018-06-07:
Closed with merge of issue_62.

closed 2018-06-07 by davidpablocohn