OceanDataTools/openrvdas #64: graceful shutdown of logger_server.

webbpinner, 2018-05-30:
This applies to the issue_62 branch.

Doing a ^C when running logging_manager currently throws an KeyboardInterrupt traceback.  If you follow this with the "quit" command the program stops responding.  ^C must be applied again to exit the program.

Doing a ^D throws an EOFError and the program immediately exits.

As ^C and ^D are commonly used as ways to exit a python script it's probably worth redirectly these commands to use the same path used for the graceful shutdown.

davidpablocohn, 2018-05-31:
Yes, this has actually been the problem that's been most dogging me with finishing up #62 - cleanly propagating the "exit" signal to all the threads and subprocesses has been surprisingly difficult.

davidpablocohn, 2018-06-07:
Forgot to tag this on the commit that fixed it in issue_62 - the trick was keeping the input() thread as the main thread and making the rest daemons.

closed 2018-06-07 by davidpablocohn