OceanDataTools/openrvdas #81: Feature Request - Serivce for retrieving data stream descriptions

webbpinner, 2018-11-07:
Add the API calls to retrieve the a description of the data format of any data stream.
The call return should be machine-readable (i.e. JSON) and include the name and UOM (unit of measure) for each column in the data stream.

This may require defining and associating a data stream format descriptor to each logger and/or logger_config.

davidpablocohn, 2019-06-27:
This is something that the CachedDataServer should be able to do. It keeps a map of field:values it has knowledge of. By initializing it with a pointer to device definition files (e.g. local/devices/*.yaml, etc), it would have access to the units, field descriptions and source device definitions for each of those fields.

Alternatively, the RecordParser could be recoded so that, whenever it outputs a field name for the first time, it also passes along that field's units and description as metadata. We could count on (hope for) the CachedDataServer picking that up and caching it as well.

The former requires more setup/configuration, but is less likely to miss things. Discuss?

davidpablocohn, 2019-09-09:
Implementation: if a RecordParser is instantiated with argument _metadata\_interval_ as not None, then for every variable it emits, every _metadata\_interval_ seconds it will add a ``metadata`` key to its output with a description of what it knows about that variable:

```
"metadata":{
    "fields":{
        "S330Roll": {"device": "s330",
                              "device_type": "Seapath330", 
                              "device_type_field": "Roll",
                              "units": "degrees",
                              "description": "Roll, port side up is positive"}, 
        "S330Pitch": {"device": "s330",
                                "device_type": "Seapath330",
                                "device_type_field": "Pitch",
                                "units": "degrees",
                                "description": "Roll, bow up is positive"}
    }
}
```
The CachedDataServer will store these and serve them when requested:
```
   {'type':'describe',
    'fields':['field_1', 'field_2', 'field_3']}
```
If 'fields' is omitted, it will return a dict of metadata for *all* fields

commit 7d1946e3 (2019-09-09): RecordParser now sends along field/variable metadata when requested. CachedDataServer stores and serves it on request. Resolves #81.
closed 2019-09-09 by davidpablocohn