OceanDataTools/openrvdas #82: Add ability to read YAML configuration files

davidpablocohn, 2018-11-10:
As Jon Meyer points out:

You really just need a dict in Python, right? While I don't know the code well enough at this point, my instinct on a migration strategy would be to augment the use of YAML while still taking in JSON.  You can do this with a conditional search for YAML, then use JSON as part of some function you can reference in all the code...

import json
import yaml

def get_openrvdas_config(basepath)
   '''Open provided path, first searching for YAML, then JSON files'''
    try:
        with open('%s.yaml' % basepath,'r') as filename:
            mydict = yaml.load(filename)
    except IOError:
        with open('%s.json' % basepath,'r') as filename:
            mydict = json.load(filename)
    return mydict

webbpinner, 2018-11-10:
I’m in favor for whatever the best format is for config files but can you or Jon make a case for YAML over JSON?  In my experience JSON is easier to deal with when using python and Javascript.

-W

=-------------------------------------------------=
Webb Pinner
Mobile: 401.749.9322
Email: webbpinner@gmail.com
Website: http://www.oceandatarat.org

davidpablocohn, 2018-11-10:
On one hand, there's practically no overhead to accepting both, as YAML can be expanded out to JSON with a line of Python. So at the very least, the path could be to treat YAML as a configuration specification language that is "compiled" down to JSON before being read in.

But for me, the most compelling upside is that, the JSON files are extremely verbose (and don't support comments), so to support concise and legible config files, I had resorted to a templated form of pseudo-JSON supported comments that could be expanded out: https://github.com/davidpablocohn/openrvdas/blob/master/docs/configuration_files.md#templates-and-variables. 

My limited understanding of YAML is that it has much of that same templating functionality built in. It's certainly more robust, better documented and more widely understood than my templating language (which even I only understand intermittently :). So it seems like a solid win that would let us throw away custom code in favor of widely-used code.

webbpinner, 2018-11-10:
I’m fine with supporting both but continuing to support JSON is a “must-have” for me.

And here’s why:
OpenRVDAS needs a GUI.  Not just a GUI to view widgets but a GUI for building/editing config files.  It is my opinion the requiring operators to open a text-editor every time a config change is required will be a deal-killer for a lot of ships… especially if new configs are required for each cruise. Sidebar… the current score is 3-1 in favor of dropping cruise configs :-).

TO EVERYONE ELSE ON THIS LIST…
Decisions are being made could have big impacts to how this project matures.  If you’ve got an opinion or just a question please share it!

- Webb

=-------------------------------------------------=
Webb Pinner
Mobile: 401.749.9322
Email: webbpinner@gmail.com
Website: http://www.oceandatarat.org

davidpablocohn, 2018-11-19:
So - after spending a day on coding with YAML, I'm converging on believing an opinion I've come across a couple of places online: JSON for serialization, YAML for configuration. Since YAML 1.2 is a strict superset of JSON, I can simply expect YAML for configurations. The eagerly-hoped-for GUI-based configuration builders can still go ahead and output JSON, and everything will be able to read it.

I'm about to check in a working branch ('yaml') that converts the documentation and nomenclature over to YAML, and gets rid of the pseudo-JSON syntax I'd adopted of using JS-style '//' comments that get parsed out on read.

davidpablocohn, 2018-11-20:
Resolved by merging branch yaml into master with commit 06de96f8eeb4209f8995b7e0472ca6a6a342906f

closed 2018-11-20 by davidpablocohn