OceanDataTools/openrvdas #93: Explore containerized/docker/etc deployment

davidpablocohn, 2019-01-18:
Current deployment options are either operating on native OS or running in a VM. I've heard from a few folks that we ought to make containerized deployments of the system, but I no pretty much nothing about containers. Would be great if someone could explore how difficult it would be to do that.

karl-malakoff, 2019-02-12:
We use virtual machines for almost everything on the RV Investigator and have also looked at Docker in the past.
 
Docker certainly has some plus points. It’s a great way to ensure you have a consistent software environment across different installations and implicitly documents the software requirements as part of the container description.
 
On the other side there is a non-trivial administrative overhead to use Docker containers. They are more conceptually complicated than VMs and a new set of commands and admin knowledge is required to use and administer them. If they have already been setup and it’s a matter of bringing them up and down it’s not so bad but getting them setup in the first place requires a lot of extra knowledge. It also relies heavily on having a working fast internet connection to pull down containers which as we know is rarely the case at sea. 
 
VMs are comparatively very easy to administer. My personal preference would be to have an OVA with a pre-built system. This can just as easily be run on a workstation for testing or on an ESXI cluster for deployment. Jules supplies an OVA for her CODAS ADCP processing system and this works extremely well. 
 
As an aside my experiments in the past have shown that UDP multicast is very problematic with Docker. Since most of our data acquisition on the Investigator uses UDP multicast I don’t think we would ever be able to use a Docker version of OpenRVDAS on the Investigator.
 
I personally think an OVA would be a much better fit that suits the most infrastructure.

davidpablocohn, 2021-11-03:
reopened, following feature request from potential user.