OceanDataTools/openrvdas #95: NMEA checksum filter transform

davidpablocohn, 2019-02-14:
Discard NMEA records that don't pass checksum test (and complain).

webbpinner, 2019-02-15:
I'm on the fence with this.  I see the value but from a purely data acquisition standpoint I do not think this is a good idea.  The purpose of the checksum is to help NMEA data consumers (auto-pilots, chart plotters, radars, etc) to ignore bad data.  In my opinion the DAS shouldn't be as picky, record the data to file but as you noted should complain so that the tech is aware of a problem.

If you work through the reverse scenario where the DAS throws out malformed data then if a problem arises (line noise, bad connectors, etc) the tech would not be able to use the recorded data to figure out what is happening.  It was this exact scenario (being able to review recorded bad data) that allowed me to figure out the baud speed mismatch error on the Falkor.

davidpablocohn, 2019-02-15:
Ah - I don't see it as something that's applied right out of the gate,
rather as something that's used further down the pipeline to help isolate
the nature of errors. E.g. - if there's a bad parse, right now you have to
look at the entrails to figure out whether it's an unrecognized record type
or a garbled record. You could insert the filter to tease those two apart:

            ____ file
raw data --/                                  ____ database
           \____checksum filter____parser____/
                      \_errors      \_errors \____ display

davidpablocohn, 2020-04-06:
Issue #207 is a duplicate of this. Its notes say:

A NMEAChecksumTransform should perform a check on the integrity/completeness of the NMEA record in question. What it outputs should be configurable at initialization. Possible behaviors include:

If checksum matches, emit record, otherwise None (to be used inline)
if checksum doesn't match, emit fail_message: a pre-specified error message (to be used on a side branch)
initialize with two optional writers as arguments - pass_writer: the writer to send the record to if it does match, and fail_writer: the one to send it to (with an optional error message) if it doesn't match
The latter is more involved but, in the end, more efficient computationally, and will result in more compact configuration files.

davidpablocohn, 2020-06-10:
Latest thoughts:

1.0
- if checksum passes, return original record
- if checksum fails, return None and logging.warning('Bad checksum for record: %')

1.1
__init__(self, error_message=None)
- if checksum fails, return None and logging.warning(error_message + record)

1.2
__init__(self, error_message=None, error_writer=None)
- allow user to pass in an optional writer and optional error message
- if checksum fails, send error message+bad record to writer

closed 2020-06-21 by anshika-agarwal