OceanDataTools/openrvdas #100: Redis-based LoggerManager backing store

davidpablocohn, 2019-03-11:
**Is your feature request related to a problem? Please describe.**
The LoggerManager keeps track of desired/current state of loggers via a backing store. Currently there are three ways it can do that: in-memory (which gets lost when LM is restarted), piggybacking on Django's backing store, or via HAPI. Given that a lot of people seem to recommend Redis for storing/passing information around (and proportionally *don't* like the weight of Django), it may make sense to have a Redis-based backing store.

**Describe the solution you'd like**
Presumably an additional accepted value for the logger_manager.py "--database" flag. Would need to specify where the server in question was, so maybe something like:
```
   logger_manager.py --websocket :8765 --database redis@localhost:6379
```
or have a separate flag for additional information:
```
   logger_manager.py --websocket :8765 --database redis --redis_server localhost:6379
```
Flag could be optional, e.g. if connecting to default hostname:port of localhost:6379

**Describe alternatives you've considered**
Might it make sense to just fold this into the HAPI interface? I really still don't understand HAPI.

**Additional context**
Add any other context or screenshots about the feature request here.

webbpinner, 2019-03-11:
where does ./database/setting.py come into play?   Does just saying '--database' and  having logger_manager use ./database/settings.py work?  This would allow for a greater number of database customizations without over complicating the cli.  Optionally the user could include the path to a custom settings.py file

i.e.
logger_manager.py --websocket :8765 --database <-- uses ./database/setting.py
logger_manager.py --websocket :8765 --database ~/customDB_settings.py <-- uses  ~/customDB_settings.py

Maybe I'm missing something.
-W

davidpablocohn, 2019-03-11:
Ah, yeah - I'm realizing that this is an unfortunate overloading of terms. The database/ dir and its settings.py file are only used by  the DatabaseWriter and DatabaseReader. The flag on logger_manager.py should really be called something like '--backing_store' instead of '--database', because it specifies what backing store the logger manager should use to keep track of what's running and what's not (in-memory, django, HAPI,...)

webbpinner, 2019-03-11:
ok... then I vote to call it "store" and use the first option
`logger_manager.py --websocket :8765 --store redis@localhost:6379`

davidpablocohn, 2019-03-11:
Sounds reasonable!