OceanDataTools/openrvdas #106: serial_reader should handle various EOL chars

julianrace, 2019-05-02:
The serial_reader script should handle EOL chars other than just CRLF.   There should be an option to specify different EOL character(s)

The eol parameter for readline() is no longer supported when pySerial is run with newer Python versions (V2.6+) where the module io is available.
The docs suggest writing your own readline method

Here's a potential solution:

`  def modifiedreadline(self):
    eol = b'\r'
    leneol = len(eol)
    line = bytearray()
    while True:
        c = self.serial.read(1)
        if c:
            line += c
            if line[-leneol:] == eol:
                #print(line.decode('utf-8').rstrip())
                break
        else:
            break
    return (line)

def read(self):
    try:
      if self.max_bytes:
        record = self.serial.read(self.max_bytes)
      else:
         #record = self.ser_io.readline()
        record = self.modifiedreadline()
      if not record:
        return None
      return record.decode('utf-8').rstrip().lstrip() #added lstrip to handle erroneous LFs
    except serial.serialutil.SerialException as e:
      logging.error(str(e))
      return None
    except KeyboardInterrupt as e:
      raise e`

**Additional context**
Some instruments output CR only instead of CRLF   (For example, Atlas-Scientific EZO circuits)

davidpablocohn, 2019-06-27:
I believe that a200521 fixes this

davidpablocohn, 2019-08-10:
Turns out I had not correctly implemented this: read_until() expects a byte object and hangs if you hand it a '\r', which is a string.

commit 52c711eb (2019-08-10): Fix handling of SerialReader eol character. We'd been handing it a string, when it wants a byte. Resolves #106
closed 2019-08-10 by davidpablocohn