OceanDataTools/openrvdas #107: Add ability to send arbitrary text to a reader when opening the socket

julianrace, 2019-05-06:
Some loggers require you to send an initialization command in order to start automatically sending data.   

You should be able to pass either a string or a file containing commands to be sent to the instrument when logging is started, and have the results recorded to a log.

For example, an SBE48 hull temperature sensor, to start up we might send a DS command, read the sensor status to a log file, and then start logging:

```
S>DS
SBE 48 V 2.0a   SERIAL NO. 0055    06 May 2019  03:59:16
battery voltage = 10.6
133 = unknown status
sample interval = 275 seconds
number of samples to average = 1
samplenumber = 4112384, free = 587483
serial sync mode disabled
real-time output enabled
SBE 39 configuration = temperature only
binary upload does not include time
temperature = 26.15 deg C
S>Interval=0
S>StartNow
S>
# 26.1487, 06 May 2019, 04:00:09
```

davidpablocohn, 2019-05-27:
This is now (in theory) covered by PolledSerialReader, though it's still untested. Julian - if/when you get the chance, could you try this in the field?

davidpablocohn, 2019-06-27:
I believe this is handled with PolledSerialReader (a200521)

closed 2019-06-27 by davidpablocohn