OceanDataTools/openrvdas #112: NetworkWriter should allow argument to bind UDP to a specific network

julianrace, 2019-05-19:
Allow optional argument to specify network adapter to send UDP datagrams out to.

Right before you do socket.connect you could call
self.socket.bind((specific-network-address,port))

On machines with more than 1 network interface, it may be nessiscary to send UDP datagrams out to only 1 specific network adapter.

davidpablocohn, 2019-05-20:
Sounds like a fine idea. Thanks, Julian!

But hmm - currently, if you specify a host/IP address along with a port
when invoking the listener (e.g. --network 192.168.56.122:6224), the
NetworkReader will try to establish a TCP connection. But no one's actually
ever asked for TCP functionality. Would it make sense to redefine the
semantics to mean that it should specifically bind to the 192.168.56.122
interface? Or should we leave the semantics as they are and throw in a new,
optional --network_interface argument?

Could you let me know your thoughts?

webbpinner, 2019-05-21:
I could be wrong but what I think Julian's asking for is the ability to restrict UDP broadcasts to a single NIC on the host (i.e. the machine running OpenRVDAS).  I agree with this request.  If the OpenRVDAS machine is dual-homed it could be useful/necessary to ensure that the UDP broadcast are only sent over one network connection.  In this case it may be more useful to bind a networkWriter to the NIC (en0, eth0, etc).

davidpablocohn, 2019-05-21:
Yes - that's what I understand, and I think it's a good idea, too (though
in the example I used the IP address rather than the name of the interface,
like eth0, etc). And - oops, I just realized the example I used was wrong -
I specified a NetworkReader, not a NetworkWriter. But the question I'm
grappling with is what the UI syntax/semantics to specify this binding
should look like. Should I hijack the syntax that's currently used for
specifying a TCP connection? Right now, if you use

   logger/listener/listen.py  [reader stuff here] --write_network foo:6224

it means "open a TCP connection to host foo on port 6224 and write to it."
Would it make sense to change it to mean "bind to interface foo and write
to port 6224 on it"? The alternative that would preserve the ability to
specify TCP vs UDP would be to keep the current syntax/semantics for the
--write_network argument, but add a new one whose purpose would be to
restrict the binding:

   logger/listener/listen.py  [reader stuff here] --network_interface foo
--write_network :6224

I'm not aware of anyone actually using the TCP functionality, so don't know
if snipping it out would actually affect anyone.

webbpinner, 2019-05-21:
I vote for --bind eth0... here's why:
This is what the user is really asking... "push data out this NIC"  The IP address assign to that NIC could change.  This will be especially true for OpenRVDAS on transient systems like fly-away ROVs where the vehicle operators need to send data into the ships network but that network's ip tuples (192.168.1.xxx, 10.1.10.xxx, etc) change from ship-to-ship and the device used to connect the fly-away network to the ship network (i.e. router) relies on dhcp for it's WAN connection.

davidpablocohn, 2019-05-21:
Makes sense! Will jump on it as soon as I've got the last bugs stomped on
v0.8.0 and pushed it out.

davidpablocohn, 2019-05-21:
Here is the network_writer.py that I modded for Alucia
Changed the first part to assume ‘host’ is actually the IP addr of the interface to send out on
Also added a CRLF to the outgoing record

################################################################################
class NetworkWriter(Writer):
  """Write to network."""
  def __init__(self, network, num_retry=2):
    """
    Write text records to a network socket.

    NOTE: tcp is nominally implemented, but DOES NOT WORK!

    network      Network address to write, in host:port format (e.g.
                 'rvdas:6202'). If host is omitted (e.g. ':6202'),
                 broadcast via UDP on specified port.

    num_retry    Number of times to retry if write fails.
    """

    super().__init__(input_format=Text)

    if network.find(':') == -1:
      raise ValueError('NetworkWriter network argument must be in '
                       '\'host:port\' or \':port\' format. Found "%s"', network)
    self.network = network
    self.num_retry = num_retry

    (host, port) = network.split(':')
    port = int(port)

    # host is specified
    # modded this to assume host is actually the IP addr of the interface you wish to transmit out on#
    # changed to UDP mode to allow Alucia to send out UDP traffic on the science network
    if host:
      #self.socket = socket.socket(family=socket.AF_INET,
      #                            type=socket.SOCK_STREAM,
      #                            proto=socket.IPPROTO_TCP)
      self.socket = socket.socket(family=socket.AF_INET,
                                  type=socket.SOCK_DGRAM,
                                  proto=socket.IPPROTO_UDP)
      self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, True)
      self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, True)
      try: # Raspbian doesn't recognize SO_REUSEPORT
        self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, True)
      except AttributeError:
        logging.warning('Unable to set socket REUSEPORT; system may not support it.')
      self.socket.bind((host,port)) #added .bind to the IP addr given
      self.socket.connect(('<broadcast>', port))

    # UDP broadcast if no host specified. Note that there's some
    # dodginess I don't understand about networks: if '<broadcast>' is
    # specified, socket tries to send on *all* interfaces. if '' is
    # specified, it tries to send on *any* interface.
    else:
      host = '<broadcast>' # special code for broadcast
      self.socket = socket.socket(family=socket.AF_INET,
                                  type=socket.SOCK_DGRAM,
                                  proto=socket.IPPROTO_UDP)
      self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_BROADCAST, True)
      self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEADDR, True)
      try: # Raspbian doesn't recognize SO_REUSEPORT
        self.socket.setsockopt(socket.SOL_SOCKET, socket.SO_REUSEPORT, True)
      except AttributeError:
        logging.warning('Unable to set socket REUSEPORT; system may not support it.')

      self.socket.connect((host, port))

  ############################
  def write(self, record):
    """Write the record to the network."""

    if not record:
      return

    # If record is not a string, try converting to JSON. If we don't know
    # how, throw a hail Mary and force it into str format
    if not type(record) is str:
      if type(record) in [int, float, bool, list, dict]:
        record = json.dumps(record)
      else:
        record = str(record)
    record = record + "\r\n"

    num_tries = 0
    bytes_sent = 0
    rec_len = len(record)
    while num_tries < self.num_retry and bytes_sent < rec_len:
      bytes_sent = self.socket.send(record.encode('utf-8'))
      num_tries += 1

    logging.debug('NetworkWriter.write() wrote %d/%d bytes after %d tries',
                    bytes_sent, rec_len, num_tries)

Julian Race
Science & IT Officer
MV ALUCIA
Email: julian@oceanx.org<mailto:julian@oceanx.org>
Voip: +1 954 672 4714
Cell: +1 508 264 3812
Fax: +870 783 206 250
www.mvalucia.com<https://urldefense.proofpoint.com/v2/url?u=http-3A__www.mvalucia.com_&d=DwMGaQ&c=6d9m4hmJHq_NZhleV8upOA&r=KVV3MSXdl_EbPHiU3abjWE5UW3LO0RWhHdenPlewqpA&m=QbHtb5ghZttl7zdNf5V96cUXr0qLjcY4vNZ8wr0YLTw&s=su70g8ojvAkA6PYsZRlmyIlQSQnE8C3HxP4w8i2pdP4&e=>

davidpablocohn, 2019-05-21:
Yup - that's the alternative I was considering, thanks!

Yet another alternative might be to simply subclass NetworkReader and
NetworkWriter to UDPReader/UDPWriter with specifically this behavior, so
invocation with the listener would be be

 logger/listener/listen.py  --udp :6224  --write_udp eth1:6225  # read from
any interface, write to eth1

Existing configs (if anyone is using them) would retain their current
behavior, and the new, more explicit invocation would be distinct and
unambigous.

I actually kinda like this path. Thoughts?

commit b3e7bcc6 (2019-06-07): UDPWriter. Also adds ability to specify eol character to append to NetworkWriter and UDPWriter. Resolves #112.
closed 2019-05-31 by davidpablocohn