OceanDataTools/openrvdas #143: Writers should (optionally) be able to split their stderr analog from their stdout analog

LMG-ET, 2019-08-01:
When a writer has a problem (let's say a file writer runs out of disk space), it might be useful to (optionally) publish that.

Maybe a CLA --stderr that you would then elaborate
Example:
   --stderr udp 6226
   --stderr :error_box.localdomain:6225

Alternatives are to just simply log the error to a file, the processes's stderr, and/or syslog

davidpablocohn, 2019-08-02:
I *think* we've got that already. One of the fields a logger config takes is "stderr_writers", which allows a logger to say where its stderr should go. E.g. (from local/lmg/monitor_and_log.yaml), the ``ladc->net`` logger has:

```
    stderr_writers:          # Turn stderr into DASRecord, broadcast to cache
    - class: ComposedWriter  # UDP port for CachedDataServer to pick up.
      kwargs:
        transforms:
        - class: ToDASRecordTransform
          kwargs:
            field_name: 'stderr:logger:ladc'
        writers:
          class: UDPWriter
          kwargs:
            port: 6225
            interface: 157.132.133.194
```
which encapsulates its stderr in a DASRecord and sends it out on UDP for the CachedDataServer to pick up. (I suppose it ought also write the stderr to a logfile, though...)

LMG-ET, 2019-08-02:
Great!  Didn’t see that in the docs, I guess I should have gone code-diving.

Guess you can mark that as “”, then.  J

 

--k

closed 2019-08-02 by davidpablocohn