OceanDataTools/openrvdas #146: Identify time gaps in the OpenRVDAS data

usapvessels, 2019-08-03:
Per discussion on LMG1906 after an interruption in the data collection, we would like to see the capability of identifying and characterizing interruptions in the data collection

webbpinner, 2019-08-03:
Would this request be more appropriately addressed in a daily QA task
performed by a stand-alone process vs in Realtime?  R2r already has
serveral QA processes that they use post-cruise for QA.  Perhaps the easier
solution to this problem would be to perform their QA checks on a regular
basis at-sea.

-W

davidpablocohn, 2019-08-04:
Definitely want to do it in daily QA as well, but I'm pretty confident that it can be added to live processing with something like a TimestampGapTransform() that would just examine timestamps of the records it receives and only emits a record if there's a gap larger than N.

webbpinner, 2019-08-04:
Ok, I need to learn more about how RVDAS can do these things.  For historical reasons I’m always going to vote to limit RVDAS to data collection, real-time transformations and real-time QA.  I’ve just seen too many good programs ruined by job creep.

Cheers,
-Webb

=-------------------------------------------------=
Webb Pinner
Mobile: 401.749.9322
Email: webbpinner@gmail.com
Website: http://www.oceandatarat.org

closed 2025-11-23 by davidpablocohn