OceanDataTools/openrvdas #147: Mysql working very too hard.

LMG-ET, 2019-08-05:
**Describe the bug**
A clear and concise description of what the bug is.

On the LMG, Django is making over 300 queries to mysql per second, resulting in unnecessarily large CPU load and IO burden. 

**To Reproduce**
Steps to reproduce the behavior:
Start logging on the LMG (running lmg_test branch of repo on Red Hat Enterprise Linux Server release 7.6 (Maipo).
Verify mysql activity using:
mysqladmin -u <openrvdas user> -p<openrvdas mysql password> status

**Expected behavior**
To provide status information to the listeners and Django GUI, only a small (configurable?) number per second queries should be required (request status of all loggers at once, parse and distribute information).

**Screenshots**
**mysqladmin:**
[image]
**top**:
[image]
**iotop**:
[image]

**Additional context**
Using lmg_test branch of GitHub repo.

davidpablocohn, 2019-08-05:
Note to self: this is because the api.update_status() (which is called once per second) method is hopelessly naive, and calls a plethora of other api methods, each with their own MySQL calls, to build its status report.

davidpablocohn, 2019-09-03:
Cached django_server_api appears to have brought this to heel.

closed 2019-09-03 by davidpablocohn