OceanDataTools/openrvdas #153: Show running config

julianrace, 2019-09-02:
Add a page to the dashboard showing the full current running config

davidpablocohn, 2019-09-02:
Do you mean being able to see the full definition for a logger's current
configuration? I've been thinking we should have something like that when
you select a logger. The pop up that lets you select/change a logger's
config should also show you the full definition for that config and maybe
even give you a check box where you can monitor that logger's output. (Of
course the latter is complicated by the fact that, what the logger outputs,
and to where it outputs it depends on the configuration...)

julianrace, 2019-09-02:
Yes, I want to be able to at a glance see what the current configuration is for each logger, including outputs.
I.E.  I want to see what UDP outputs the GPS is currently set up to deliver.

davidpablocohn, 2019-09-27:
Now fixed in dev branch. Will be merged into master at next version

commit 0185d26e (2019-10-19): Display collapsible version of current config JSON when selecting logger configs. Resolves #153.
closed 2019-09-29 by davidpablocohn