OceanDataTools/openrvdas #163: Format the content of server messages send via websocket as JSON instead of a single string.

webbpinner, 2019-10-01:
Currently server log messages are broadcasted to connected web-client in the following format:

`{"type": "data", "status": 200, "data": {"stderr:logger_manager": [[1569923851.814403, "30\tWARNING\t2019-10-01T09:57:31.814Z cached_data_writer.py:98: CachedDataWriter websocket connection to localhost:8766 failed; sleeping before trying again: Multiple exceptions: [Errno 61] Connect call failed ('::1', 8766, 0, 0), [Errno 61] Connect call failed ('127.0.0.1', 8766)"], [1569923852.995031, "30\tWARNING\t2019-10-01T09:57:32.994Z logger_runner.py:444: Process for pco2 (config pco2->net) unexpectedly dead; restarting"], [1569923853.005235, "30\tWARNING\t2019-10-01T09:57:33.004Z logger_runner.py:444: Process for tsg2 (config tsg2->net) unexpectedly dead; restarting"], [1569923853.012958, "30\tWARNING\t2019-10-01T09:57:33.012Z logger_runner.py:444: Process for pguv (config pguv->net) unexpectedly dead; restarting"], [1569923853.028529, "30\tWARNING\t2019-10-01T09:57:33.027Z logger_runner.py:444: Process for PCOD (config PCOD->net) unexpectedly dead; restarting"]]}}
`
This works well for the current web-interface but limits the usefulness of the message by other clients because the messages are baked in to preformatted string.

Ideally I feel it would be great to have the pre-baked string replaced with a JSON-formatted object that could be easily parsed by other clients.

I.e. replace this:
[
  1569923853.248462,
  "30\tWARNING\t2019-10-01T09:57:33.247Z django_server_api.py:632: message_log() Got DjangoOperationalError. Trying again: (2006, 'MySQL server has gone away')"
]

with:
{
  timestamp: 1569923853.248462,
  log_level: 30,
  message: "django_server_api.py:632 message_log() Got DjangoOperationalError. Trying again: (2006, 'MySQL server has gone away')"
}

As an aside... here is the structure of adding a message to the data store via the API.
API_LOGGING_FORMAT = '%(filename)s:%(lineno)d %(message)s'
self.api.message_log(source='Logger', user='(%s@%s)' % (USER, HOSTNAME),
  log_level=record.levelno,
  message=self.formatter.format(record))

Unifying these two formats would be another nice to have polish.

davidpablocohn, 2019-10-02:
Oh yeah - this makes perfect sense. Thanks!

davidpablocohn, 2019-11-20:
Integrated in branch issue_172, merged to v0.9.2

closed 2019-11-20 by davidpablocohn