OceanDataTools/openrvdas #166: Non-Highcharts line widget

davidpablocohn, 2019-10-05:
The Highcharts line widgets are great, and free licenses are available for academic institutions,  but they are not open source.

MignonBelongie, 2019-10-14:
I can take a shot at a D3 based TimelineWidget. I don't have any prior experience with D3, but I've been playing around with it for a few days, and I now have a very rough implementation of a TimelineWidget working with WidgetServer.

vwarner1411, 2019-10-15:
I wrote a bunch of D3 based stuff for the das a few years back. It should
all exist still. I can dig it out if it’s not on my github. @vwarner1411

MignonBelongie, 2019-10-15:
Thanks, I'll look at it. I actually came across some helpful code of yours today in stackoverflow, too.

davidpablocohn, 2019-10-15:
That would be great - thank you!

davidpablocohn, 2019-10-23:
At RVTEC - several people suggested looking at Rickshaw (https://tech.shutterstock.com/rickshaw/examples/), which is open source, apparently built on top of D3. Might be worth a look.

MignonBelongie, 2019-10-24:
I have something ready to demo, but I just tried pushing, and it looks like I don't have permission any more.

davidpablocohn, 2019-10-24:
I just tried adding you. Can you try sending a pull request?

closed 2019-12-09 by davidpablocohn