OceanDataTools/openrvdas #175: D3 TimelineWidget needs work

MignonBelongie, 2019-11-05:
See discussion in https://github.com/OceanDataTools/openrvdas/pull/171 about jitter.

Also, some options were requested in https://github.com/OceanDataTools/openrvdas/pull/173:
 - min, max  - optional limits on max min display values
 - minRange - minimum difference between max and min

davidpablocohn, 2019-12-05:
Many thanks for your work on this so far!

First pass at replicating NBP demo dashboard with D3 widgets ([http://165.227.31.102:8000/display/nbp_dashboard_d3.html](http://165.227.31.102:8000/display/nbp_dashboard_d3.html), or see [https://photos.app.goo.gl/W8pmR4gK83rLVXmf7](https://photos.app.goo.gl/W8pmR4gK83rLVXmf7)). Functionality is there, but lots of little issues affecting usability:
 - x-axis ticks seem seem to be per-point?
 - can we control number number of major ticks on Y axis?
 - control font size of x axis label?
 - set background color?

I expect these can be controlled via CSS, but don't understand D3 well enough to figure it out. Also worried about the pure CSS approach (vs widgetOptions) to the extent that it feels like it applies to all widgets of a type, not letting us have different widgets of the same type have different appearances. Is there a workaround to this?

davidpablocohn, 2020-08-14:
I think we're going to deprioritize this for now.

closed 2020-08-14 by davidpablocohn