OceanDataTools/openrvdas #194: Expand install instructions to explain how to run OpenRVDAS in virtual python environment

webbpinner, 2020-01-19:
Python is a complicated beast with the potential for conflicting dependencies.  One proven way to combat this is to use virtual python environments that allow applications to essentially run in a clean python installation.

This is how I currently run OpenRVDAS on my Mac laptop.

I do not envision or recommend the use of virtual python environments for use in production environments.  This suggestion is simply to help new users get a test platform up and running.

davidpablocohn, 2020-01-22:
Seems like an excellent idea. One thing that gives me pause is that it looks like starting processes with supervisord is a bit convoluted under virtual environments - do you have any special tricks for that?

davidpablocohn, 2020-01-22:
The Cybersecurity folks gave a talk at RVTEC in Fairbanks and came to my OpenRVDAS preso. I recall them encouraging me to move to a virtualenv installation, but I didn't press for details. What's your thinking on avoiding venv in production? (I've figured out the whole supervisord + venv thing, and it's pretty straightforward)

webbpinner, 2020-01-23:
I've been a little under the weather this week and slow to respond.

Production environments for most vessels will most likely be dedicated hardware (or at least IMO it should be).  In those instances a venv just complicates the install with an additional layer of abstraction.  I see the real benefit to venv as helping folks wanted to test-drive OpenRVDAS on their laptops or a development server.

davidpablocohn, 2020-02-02:
Resolved by #120

closed 2020-02-02 by davidpablocohn