OceanDataTools/openrvdas #202: Move InfluxDB settings into database/ subdir, expand instructions for use

davidpablocohn, 2020-02-24:
Also create (commented out) supervisor command block to run InfluxDB.

webbpinner, 2020-02-27:
I'd like to discuss this as it pertains to the development of future readers/writers/transforms.

If MySQL is going to be come optional (#204) then is it intuitive to the end-user to include InfluxDB settings in the database settings file?

I'm wondering if perhaps there needs to be a higher-level nomenclature, schema, file hierarchy, etc for how reader, transforms, writer specific settings are organized (when required).

Maybe have a settings folder?  I'm open to ideas.  Given the overall modular natural of OpenRVDAS I'm not opposed to having the settings spread across multiple module-specific files.

davidpablocohn, 2020-02-27:
Yup - this is definitely worth discussing to make sure we get it right. Can
we schedule a call to talk? I'm generally free except for this weekend.

davidpablocohn, 2020-03-21:
I'm now thinking, given the positive feedback we've received about working with InfluxDB, that we should incorporate it into the build script as an optional component, the way we've done with MySQL: "Install InfluxDB?" (yes/no) "Run InfluxDB as a service on startup?" (yes/no)

commit 6cf613e4 (2020-03-24): Update CentOS script for InfluxDB. Also lower warning level if databases aren't configured. Resolves #202.
closed 2020-03-25 by davidpablocohn