OceanDataTools/openrvdas #203: Interpreted Data Writer

mchoi39, 2020-02-26:
**Is your feature request related to a problem? Please describe.**
I want to be able to write the interpreted data that openrvdas creates into a file, not just the sentences it parses

**Describe the solution you'd like**
A file writer that writes the data that was extracted. For example, if it extracts wind speed and the value is 5kph, I want to be able to save that into a file

**Describe alternatives you've considered**
There are no other ways as of now

davidpablocohn, 2020-02-26:
Hi Mateo - I've started a feature request here: #201 - for something that I think does what you're asking. Could you have a look and let me know if that's what you're after?

mchoi39, 2020-02-26:
Hi Pablo,

Yes this seems to be what I'm looking for.

davidpablocohn, 2020-03-17:
Resolved along with #201 in latest dev branch push by implementation of FormatTransform

closed 2020-03-17 by davidpablocohn