OceanDataTools/openrvdas #206: RecordParser should properly parse NMEA-style lat/lon formats

davidpablocohn, 2020-04-03:
NMEA-formatted lat/lon values require conversion. In addition to the existing custom formats like :ow and :of (which should be broken out into a separate file), we should have :nlat and :nlon or something like that that takes the raw number and converts it into decimal degrees (as is used, e.g., by Leaflet and some other mapping systems).

From StackOverflow:

The format for NMEA coordinates is (d)ddmm.mmmm
d=degrees and m=minutes
There are 60 minutes in a degree so divide the minutes by 60 and add that to the degrees.

For the Latitude=35.15 N
35.15/60 = .5858 N

For the Longitude= 12849.52 E,
128+ 49.52/60 = 128.825333 E

In php, you could do this:

<?php
$lng = "12849.52 W";

$brk = strpos($lng,".") - 2;
if($brk < 0){ $brk = 0; }

$minutes = substr($lng, $brk);
$degrees = substr($lng, 0,$brk);

$newLng = $degrees + $minutes/60;

if(stristr($lng,"W")){
    $newLng = -1 * $newLng;
}

?>

davidpablocohn, 2020-04-04:
Question: should we also parse out the E-W/N-S component? As far as I can see, the numerical part of an NMEA lat/lon is always followed by a comma and the N/S/E/W character. So we could parse that out as well, and convert it to a full +/- value.

Disadvantage is that some applications may want the N/S/E/W character for display purposes.

Alternative would be to provide two additional formats: one that just parses/converts the numerical part, and one that parses/converts the number plus direction. Thoughts?

webbpinner, 2020-04-04:
If the parser is making latitude in the Southern Hemisphere and longitude in the Western Hemisphere negative then application needing a "N/E/S/W" designation can just do a >0 comparison.

davidpablocohn, 2020-04-05:
Okay - I've added two formats: nlat (for converting just the NMEA "dddmm.mmmm", and nlat_dir for 'dddmm.mmmm,[NEWS].

davidpablocohn, 2020-04-05:
Resolved in dev branch.

commit 82e01164 (2020-04-05): Add NMEA-format parsing for lat/lon. Also add check_parse_format.py script. Resolves #206. Resolves #209.
closed 2020-04-05 by davidpablocohn