OceanDataTools/openrvdas #245: Unify installation scripts into single script

davidpablocohn, 2020-09-16:
Right now, we've got three arch-dependent installations scripts for OpenRVDAS: build_openrvdas_[centos7,ubuntu18,macos]. The install_influxdb script is an example of detecting the installed distribution/OS and doing the right thing. It would be desirable to combine the three build_* scripts into a single install_openrvdas.sh script that operates the same way.

webbpinner, 2020-09-16:
while not the same as a unified script... what about using the system's package managers (yarn, apt, brew)??

davidpablocohn, 2020-09-16:
We do that inside the script, but each system uses a different package
manager, and some of the things we need to install don’t have packages
already built in the versions we need. Which is what makes each install
script complicated. For example, CentOS7 Doesn’t have the right version of
python or Django, so we have to install those separately.

davidpablocohn, 2020-10-07:
Note: while doing this, also move Python packages into a requirements.txt file

davidpablocohn, 2020-11-11:
Merged to dev.

closed 2020-11-11 by davidpablocohn