OceanDataTools/openrvdas #248: RecordParser doesn't capture NMEA message type

davidpablocohn, 2020-10-25:
**Describe the bug**
When the old NMEA parser encountered a record such as 
```
s330 2017-11-04T05:12:19.479303Z $GPZDA,000000.17,07,08,2014,,*78
```
it captured `data_id=s330` and `message_type=INZDA`. The new RecordParser doesn't capture `message_type`, and its current format syntax doesn't support any way to specify what a record's message type is:
```
  format:
    - "$GPGLL,{Latitude:nlat},{NorS:w},{Longitude:nlat},{EorW:w}"
    - "$GPVTG,{CourseTrue:of},T,{CourseMag:of},M,{SpeedKt:of},N,{SpeedKm:of},K"
    - "$GPZDA,{Time:d},{Day:d},{Month:d},{Year:d},{LocalHours:od}"
    - "$GPRMC,{Time:f},{Status:w},{Latitude:nlat},{NorS:w},{Longitude:nlat},{EorW:w},{SpeedKt:f},{Cou
rseTrue:f},{Date:w},{MagneticVar:of},{MagneticVarEorW:ow},{Mode:w}*{Checksum:x}"
```

**Recommendation**
Alter RecordParser to accept an enhanced syntax: instead of just accepting a list of strs, allow it to accept a list of strs or a dict where the key is a message_type str and the values are either strs or lists strs corresponding to formats that match that message type:
```
  format:
    GLL:
    - "$GPGLL,{Latitude:nlat},{NorS:w},{Longitude:nlat},{EorW:w}"
    - "$INGLL,{Latitude:nlat},{NorS:w},{Longitude:nlat},{EorW:w}"
    VTG:
    - "$GPVTG,{CourseTrue:of},T,{CourseMag:of},M,{SpeedKt:of},N,{SpeedKm:of},K"
    - "$INVTG,{CourseTrue:of},T,{CourseMag:of},M,{SpeedKt:of},N,{SpeedKm:of},K"
    ...
```

webbpinner, 2020-10-25:
I believe the NMEA 0183 spec defines the the message identifier as starting with $, allowing for up to 2 user-defined characters and then the NMEA defined message id (GGA, GLL,ZDA, VTG, etc).

Take a look at Kurts post from 5.29.2009 where he defines the full regex pattern for a GPS sensor on the Healy
http://schwehr.org/blog/archives/2009-05.html <http://schwehr.org/blog/archives/2009-05.html>

-Webb

=-------------------------------------------------=
Webb Pinner
Mobile: 401.749.9322
Email: webbpinner@gmail.com

davidpablocohn, 2020-10-25:
Thanks - those regexes will be helpful as we add more message types.

For now, the problem is that the format in which the RecordParser ingests
formats doesn't have any way to associate a particular string with a
particular message type.

commit 73a9dcbf (2020-10-28): Add message_type ability to RecordParser.
Resolves #248.
closed 2020-10-30 by davidpablocohn