OceanDataTools/openrvdas #271: Discussion: break org/vessel-specific code/data into separate repos

davidpablocohn, 2021-07-28:
Right now, we're not really using branches the way Git wants us to. Officially, branches are for code that is eventually going to get merged back into the main project. But we've got usap, rcrv, uaf, etc branches that we're intending to run in parallel. The official "right way" to do that is with a fork - there should be a usap fork, rcrv fork, etc.

The problem is that for now, these orgs don't have their own GitHub accounts (or at least that we have access to), and GitHub doesn't let one have forks within a single org. So OceanDataTools can't maintain an openrvdas-usap fork.

The most straightforward way I see around this is changing how we use the local/ subdir. We already segregate org/vessel specific code there, e.g. usap code goes in local/usap, and RCRV in local/rcrv. I propose that we have org/vessel-specific code be separate repos that get checked out/in under local/. So the main openrvdas branch would have a relatively empty local/ subdir, and you'd check the openrvdas-usap-local repo out into into it to populate it with USAP code.

Thoughts?

KPed-at-home, 2021-08-02:
OK... makes sense from a GitHub standpoint.   I personally like that with a fully populated "local" we get to see what other vessels are doing with OpenRVDAS.  Brings fresh ideas.    But if you're running this on a PiZero so something, you want to stay minimal.  So maybe a provide a util script that will populate local/[x] with all the [x]-specific stuff from all the curated installations, but the default installation would be empty-ish and ask you which "other" repo [formerly known as branch] to pull in.

webbpinner, 2021-08-02:
I'm taking a similar approach with Sealog where the custom clients for the various users are branches of the main branch.  I've moving towards getting these custom branches correctly organized so that any bug-fixes made on the main branch can be pulled into the custom branches using git's rebase operation.

This doesn't preclude users from submitting their ideas/improvements/bug fixes but it may required these additions to be submitted as pull requests against the main branch.

I do agree that there is a need to further flesh out the post-install setup instructions and perhaps develop a "best practices" guide.

closed 2024-03-27 by davidpablocohn