OceanDataTools/openrvdas #277: Use python's pty library instead of `socat` for data simulation

KPed-at-home, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
No problem.

**Describe the solution you'd like**
Instead of delegating to socat for the virtual serial ports for data simulation, use python's pty library (https://docs.python.org/3.8/library/pty.html).  I just have this fetish where I think it's better to write python in python (not shell).

**Describe alternatives you've considered**
Another alternative might be to use RFC2217 serial ports (https://pyserial.readthedocs.io/en/latest/examples.html), but I've only used that from one box to another.  And it is marked as experimental, so I'd hesitate to incorporate it into a curated product.

davidpablocohn, 2021-08-03:
KPed - Thanks for all of these! At the moment I'm focusing on #283, which appears to be the biggest villain in several ways. If you've got time, would you be willing to take this one up?

KPed-at-home, 2021-08-05:
I'll look at it.  Thanks.

davidpablocohn, 2021-09-06:
Merged issue_277 into dev

commit 764dd05b (2021-09-07): Merge branch 'issue_277' into dev
Resolves #277.
commit 95eee4e2 (2021-09-07): Remove socat dependency in simulate_data.py. Resolves #277.
closed 2021-09-06 by davidpablocohn