OceanDataTools/openrvdas #278: Good way to help figure out why parsers are failing

KPed-at-home, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
I have one parser definition (for the Seapath 330) where the last definition is "{unparseable:os}", and yet OpenRVDAS tells me no pattern matches the data string.

**Describe the solution you'd like**
I'd like a utility or flag the turns on ridiculous levels of debug for the parser, where it tells me what pattern it's using, shows me something like what "check_parse_format.py" for each of them.  Seems like with a pattern that's basically a regex dot-star, I'd match ANYTHING, but I'm not matching, so... maybe there's a problem in my YAML and it's not seeing that pattern (this would tell me), or maybe I've uncovered a bug, or maybe it just doesn't like me....

**Describe alternatives you've considered**
Voodoo and tweaking the code myself are the two options I've considered (it's easy enough to futz around and then just git checkout the file you messed with)

**Additional context**
Currently it's quieted, but see LMG's /var/log/openrvdas/lsep.stderr

webbpinner, 2021-08-02:
I've had similar scenarios and used the following command to help troubleshoot:
`.venv/bin/python` logger/listener/listen.py --config_f local/usap/cruise.yaml:s330-\>file -v`

This allows you to run the individual logger config in a terminal with additional verbosity (the -v flag).  I'm sure that command isn't exactly what you need to use but hopefully you get the idea.  It needs to be run from the OpenRVDAS install root which is typically /opt/openrvdas.

Be sure the that logger is set to "off" within OpenRVDAS before running this command so that OpenRVDAS releases control of the serial/udp port

KPed-at-home, 2021-08-02:
Been there done that.  Not quite enough.  It's easy enough to mess with the python module and add the debugging I need, then just "git checkout" to clean up after my mess, but I'd like to assume that I'm not the only one having similar issues. Or maybe it *is* just me.  But obviously this type of issue was ubiquitous enough for the creation of the check_parse_format command, so.....

davidpablocohn, 2021-08-05:
So yeah - maybe something like a CheckParseFormatTransform that can drink
directly from the firehose (so you don't have to worry about chars getting
lost on the way), and outputs diagnostics on what matches, and how much?

closed 2026-09-30 by davidpablocohn-claude-bot

webbpinner, 2026-10-01:
Have you used the parser helper tool in the new ReactUI?