OceanDataTools/openrvdas #280: A non-interactive tool for changing logger state or cruise mode

KPed-at-home, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
No problem.  Enhancement request.

**Describe the solution you'd like**
Tweak to the server_api_command_line  tool so that it can accept commands on the command line rather than interactively, or ideally a simplified tool (minimal required command line options) to do the same thing.   The only use case I currently have in mind is geo-fencing, but it seems like something that might have broader application.  Like with an ROV, fire a (hypothetical) process writer to change mode from "on deck" to "deployed" when you see pressure on the depth sensor..

**Describe alternatives you've considered**
Writing an expect script to do this using existing tools.

webbpinner, 2021-08-02:
The need for a RESTful API that would allow operators to request state changes (update config, switch modes, retrieve logger config, etc) via an http GET/PUT/POST has been discussed and is planned for a future release.

For the ROV example you cited...
A similar functionality has been achieved by combining the functionality of OpenRVDAS with the other OceanDataTools.org projects (Sealog, OpenVDM).

What is done on other vessels with ROVs is to set the DAS (OpenRVDAS, SCS, etc) to always log incoming data to file.  The resulting files are regularly pulled from the DAS workstation to a centralized ship's data warehouse server by OpenVDM.

In parallel scientists and vehicle operators use Sealog to capture events from the dive (including events like "off-deck", "on bottom", "off bottom", "on deck").

At the end of the dive the vehicle operator confirms/corrects the launch/recovery times in Sealog.  Afterwards they run a script that uses the Sealog API to extract the dive's corresponding start/stop times and the OpenVDM API to extract the location of the OpenRVDAS files on the data warehouse to crop/combine the raw data files into new data files that corresponds to the exact start/stop time of the dive.

This approach has proven to be quite robust as there are no real-time constraints that would potentially lead to data loss.  Because OpenRVDAS is always running, it will always write data from the vehicle to file.  Because Sealog allows the operator to correct any mistakes in the dive milestone timestamps the is no pressure to get everything right in real-time.  Thus the cropped/combined data files for a dive can be exported, tweaked and re-exported as needed to ensure the final exported dataset is correct.

A similar approach could be used with Sealog/OpenRVDAS/OpenVDM for culling data collected within EEZs whereby a list of time periods when the vessel is within the EEZ is captured with Sealog and a script is developed that leverages the Sealog and OpenVDM API to remove EEZ data from the delivered cruise data package.

KPed-at-home, 2021-08-02:
GET/PUT/POST query and set is highly desirable, obviously.  But it assumes the inclusion of a webserver and some form of middleware (most probably Django/Flask/etc).    While it's true that currently there is no "headless" option in the installer script, I see that as only a matter of time.

webbpinner, 2021-08-02:
Using something like hapi, flask, fastAPI are their own "servers" and would not require running a traditional web-server like apache or nginx.  I can imagine that even with a headless installation there would be a desire to control OpenRVDAS regardless of if it's running on the local machine or a remote server.  This RESTful API approach would handle both scenarios with the same code.  The API would be the back-end for a cli utility.  The cli utility would have a config file that specifies the host/port of the API (localhost vs some IP) and optionally enforce role-based security policies.

davidpablocohn, 2021-08-04:
Pushed to dev, usap and usap-dev branches.

"""
server/lmcmd.py

A command line tool for interacting with the LoggerManager and database. If called
with no arguments other than the optional database specification, it starts an
iterative command line interface. If arguments are present, it treats them as a
an OpenRVDAS command line command, executes it, them, then exits.

Within the iterative command line interface, you can type 'help' to see
available commands, and use arrow keys to navigate command history.

Args:
    --database
        If specified, either 'django' or 'hapi'; which database
        to connect to for OpenRVDAS state. If not specified, defaults to
        'django'.
    -v
        If specified, increase verbosity of diagnostics from 'warning'
        to 'info'. If repeated, increase to 'debug'.

Usage:
    server/lmcmd.py
        Start a command line reader that reads/writes the default Django
        database used by OpenRVDAS.

    server/lmcmd.py help
        Print available interface commands and exit.

    server/lmcmd.py --database hapi
        As above, but use HAPI database

    server/lcmd.py get_active_mode
        Connect to default (django) database and run the 'get_active_mode'
        command, then exit

    server/lcmd.py --database hapi set_active_mode underway
        Connect to HAPI database, execute command 'set_active_mode underway',
        then exit.
"""

commit 62f3897e (2021-08-04): First pass at standalone command line interface. Resolves #280.
closed 2021-08-04 by davidpablocohn