OceanDataTools/openrvdas #281: Pure SQLITE back-end for headless deployment

KPed-at-home, 2021-08-02:
**Is your feature request related to a problem? Please describe.**
Currently you need HAPI or Django to get a persistent back end (the in-memory back end won't survive a power cycle).  A pure SQLITe back-end would allow OpenRVDAS to come back to the state it was in on power up without requiring any sort of web interface.

**Describe the solution you'd like**
The api cli is sufficient for most uses, but adding a non-GUI persistent back-end seems desireable

**Describe alternatives you've considered**
Nope... that's it

**Additional context**
Add any other context or screenshots about the feature request here.

webbpinner, 2021-08-02:
Could this be achieved by passing the config yaml file in the system init script?
i.e.`.venv/bin/python logger/listener/listen.py --config_f local/usap/cruise.yaml`

This would require that the default mode listed in the config file is the desired mode (i.e. 'underway' vs 'off').  But in a headless application there may only be the need for just an 'underway' mode.

KPed-at-home, 2021-08-02:
Ummm.... yes and no, I think.  You can get as minimal as you like, of course.  The ideal end-point of having a non-GUI backend is full OpenRVDAS functionality, just no GUI whatsoever.  Barring other magic your only control would be the server api cli tool, but ideally you'd get all the goodness OpenRVDAS has to offer, like coming up in the last commanded state after a restart/power-cycle.

FrozenGeek, 2023-06-10:
My first big OpenRVDAS mod, so may (meaning certainly) needs some work so issues *VERY* welcome, but I think this pretty much takes care of it.

As a minimal implementation, the only upstream file I modifed in my branch was server/logger_manager.py, 
but server/server_api_command_line.py and server/lmcmd.py could also be modifed to accept the SQLiteServerAPI. (just a half-dozen lines of code each)

https://github.com/FrozenGeek/openrvdas/tree/sqlite_gui/sqlite_gui

FrozenGeek, 2023-06-10:
BTW:  FrozenGeek is KPed

davidpablocohn, 2023-06-10:
Excellent - thank you! Next two days are a bit crowded, but will dive in on
Monday.

davidpablocohn, 2023-06-23:
Apologies apologies apologies! Life got away from me. FINALLY reviewing
this code.

davidpablocohn, 2023-06-23:
I've pulled your sqlite_gui branch into "issue_281" on the main repo, and
am poking at it there. Many thanks for the extensive NOTES.md!

davidpablocohn, 2023-06-23:
Looks like you miss the special case where the db is empty:

  File "/opt/openrvdas/sqlite_gui/sqlite_server_api.py", line 249, in
get_configuration

  File "/opt/openrvdas/sqlite_gui/sqlite_server_api.py", line 223, in
_do_we_need_to_reload

    self._do_we_need_to_reload()

  File "/opt/openrvdas/sqlite_gui/sqlite_server_api.py", line 223, in
_do_we_need_to_reload

    row0 = rows[0]

           ~~~~^^^

IndexError: list index out of range

    row0 = rows[0]

           ~~~~^^^

IndexError: list index out of range

Looks like a pretty simple fix - shall I go ahead and patch where I see
issues, or would you like to?

FrozenGeek, 2023-06-24:
I don't know how to add you to the project as a git-pusher to this project, so... I guess... uh... open issues so I can fix them?  I also don't know how to merge pull requests, but there are a lot of docs and guides for github online, so it seems likely we can resolve both these git problems.

But yeah, patch away.  We'll figure out how to include them either here or on your end.

As for the specific issue you've got, I had thought I'd set things up so that it checked for an empty database when it opened it.... somewhere in the process of streamlining the code I must have removed the code that made that particular problem unreachable.  

I'll open an issue for this one.

davidpablocohn, 2023-06-24:
It took me a bit of googling to figure out how to merge across forks, but
it's easy to do with pull requests. I'll file a PR to merge my changes back
into your fork, including updating it to include the latest dev changes.

FrozenGeek, 2023-06-25:
I've already got a patch there, but if it it's differennt from yours it's easy enough to back out and merge in yours.

Now I guess I need to read up on PR's :-)

davidpablocohn, 2023-06-25:
I'm still learning about PRs myself. I had hoped that I could do a pull
request from the official repo into yours (to merge in the latest "dev"
goodness). But it looks like PRs only work upstream. To set one up
requesting a merge from your repo the the main one, you go to the PR tab,
select "New Pull Request," and then the drop-downs let you select from/to.
Ideally, we'd pull your changes into openrvdas/issue_281, then once it's
all nailed down and bullet-proofed, merge it into dev.

Based on my amateur understanding of the whole process, I'm going to try to
fetch openrvdas/issue_281 into your repo (as frozengeek/issue_281). It has
your code, with dev merged in, and my three lines of patch. We can then do
all the code review-y things on it, and pull to the main repo when ready.

davidpablocohn, 2023-06-25:
Okay - I've created an issue_281 branch on your fork with my little patch,
and the dev changes merged in.

davidpablocohn, 2023-06-27:
I've gotten the gui up and running on my Ubuntu test machine, but am still
struggling with getting the cgi-bin auth to take the username and password
I created with Usertool. More tomorrow...

davidpablocohn, 2023-06-28:
Okay, I think what I'm going to focus on is munging the setup scripts -
they've got a few issues, the main one of which is that where they put
the files they generate depends on where they're invoked from. E.g. if you
run sqlite-gui/nginx/GenerateCert.sh from opt/openrvdas, it puts the certs
in /opt/openrvdas, but nginx looks for them in
/opt/openrvdas/sqlite-gui/nginx. Et cetera.

FrozenGeek, 2023-06-28:
If you use the utils/install script, it *should* work the way it's supposed to, but yeah... makes sense that it should work from anywhere.  Thanks for looking at that.

I didn't want to get too far into those scripts until I got your input on the best way to make this an add-on.

The "contrib" route seems inappropriate.  I like the way you're doing vessel repos... install the add-on where you please and symlink it into the installed openrvdas tree.  Seems like a pretty good route.  The only thing that doesn't really "fit" is the mod to logger_manager.  That I might just encourage you to apply to OceanDataTools:OpenRVDAS directly.  I mean... if someone selected "--database sqlite" without installing the addon bad things would happen, but in that case they kind of get what they deserve.

davidpablocohn, 2023-06-29:
Aha! I'd entirely missed that there was a utils/install_sqlite_gui.sh and
was doing a piece-by-piece installation based on the NOTES.md file. I'll
back out and try it the easy way. :)

davidpablocohn, 2023-07-02:
Heads up that I'm still at it. Having trouble getting
sqlite_gui/utils//install_sqlite_gui.sh to run as intended. On the road for
the next two days, will resume on Tues.

davidpablocohn, 2023-07-10:
Urgh. Merging the sqlite_gui branch deletes a mess of essential files from
the database/ subdir.

davidpablocohn, 2023-07-21:
Heads up that I actually *am* working on this. I've got a branch that I'm
modifying to enhance the installation script so that it works robustly on
MacOS. I'll file a pull request to merge it into master once I've got it
working seamlessly.

davidpablocohn, 2023-07-22:
Note: I've folded the logger_manager.py changes into dev and main branches
of openrvdas, along with an import check that only shows the 'sqlite'
option for database if sqlite_gui/ actually exists and is linked from the
openrvdas install dir.

davidpablocohn, 2023-07-27:
Kevin - unless you have any objections, I think I'd like to move
sqlite_server_api.py over to openrvdas/server (where
in_memory_server_api.py) is, and just make it installed by default as part
of the main branch. I've made a few changes to it to automatically create
and initialize the sqlite database, and to allow specifying the path where
the db should live.

I've got it in a side branch at the moment (
https://github.com/OceanDataTools/openrvdas/tree/sqlite_db), since I
couldn't figure out how to copy over the history, and I don't want the
history of your development of those files to get lost in Git.

davidpablocohn, 2023-10-24:
Now checked into master branch

closed 2023-10-24 by davidpablocohn