OceanDataTools/openrvdas #283: Fix logger_manager stderr listening

davidpablocohn, 2021-08-03:
**Describe the bug**
Right now, the logger_manager sets up threads to listen to each logger's stderr file to relay those lines to the console and cached data server. When one or more loggers is generating a lot of stderr output (say, because of a bad parse format), the logger manager gets overloaded and starts timing out on other tasks. This causes cascading failures.

The problem seems to be exacerbated when the logger_manager starts up if logger stderr files are already large.

davidpablocohn, 2021-08-04:
Something in logger_manager.py:_check_logger_stderr_loop() is gradually sucking all processing resources. This is what relays the stderr of the individual loggers off to the CDS. For now, in dev branch, I've disabled it.

KPed-at-home, 2021-08-05:
One possible solution might be to just have the stderr writer write to the CDS itself.
Thus eliminating the need to tail anything.  You could check that the CDS exists (more efficient) but you could probably just also ignore CDS-write errors.

davidpablocohn, 2021-08-18:
Fix, sending/reading stderr directly to/from CDS, checked into dev branch.

closed 2021-08-18 by davidpablocohn