OceanDataTools/openrvdas #294: Somehow encourage users toward standard variable names

davidpablocohn, 2021-11-04:
The lovely thing about standards is that there are so many of them. :) But there seem to be two main camps of variable names. Peter Shanks writes:

_"It’s a bit of a can of worms, but there’s a good argument for getting everyone to use the same names for the same things. Unfortunately there are two main camps (and a couple of outliers, but we’ll ignore those for now). CF (climate and forecast) metadata https://cfconventions.org/ and NERC (the UK’s Natural Environment Research Council) https://vocab.nerc.ac.uk/. The former tend to be used more commonly in NetCDF data and describe what’s being measured, while the NERC parameters generally describe what’s being measured but can also include which instrument was doing the measuring, and how and where the measurement was taken. A lot more informative, but can also lead to quite unmanageable variable names (my current favourite: sea_surface_secondary_swell_wave_period_at_variance_spectral_density_maximum). NERC also splits things up into collections, which can be a bit confusing if you’re coming to it cold. I’d be using CF if it came to a choice, but there are many variables that CF doesn’t cover._

_As an aside, there is a CF NERC collection: http://vocab.nerc.ac.uk/collection/P07/current/"_

https://cfconventions.org/Data/cf-standard-names/current/build/cf-standard-name-table.html

LMG-ET, 2024-03-28:
So our choices for standardization seem to be the woefully incomplete cfconventions or the absurdly verbose NERC vocabulary?

While I was changing our cruise configuration template to be more in keeping with R2R's suggested/preferred file naming convention, I looked into what suggested conventions were out there for more internal uses.  The Ocean Data Interoperability Platform looks like a dead program, the WMO's JCOMM page is really just a placeholder for broken links, and OceanBestPractices is silent on the subject.  So while standard names might be nice, I'm afraid it's mostly every man for himself.

webbpinner, 2024-03-28:
That has been my experience having worked for just about every US-based institute.  Best we can hope for is cruise-to-cruise consistency within a single institute.

LMG-ET, 2024-03-28:
Maybe a better set of comments and a full set of descriptions in the device type examples would encourage people to make use of those, and at least for things like GPS's (where we know every ship's DAS is collecting at least one) we could get some sort of standarization... peeps is peeps, they're just going to /bin/cp the files or cut/paste.  If the defaults are good....

davidpablocohn, 2024-03-28:
We had a good conversation about this in Hobart and kept coming back to:
https://xkcd.com/927/ - none of the existing naming conventions out there
seem to cover everyones' needs. The phrase that we ended up with (was this
Webb's phrase?) was "consistency, not conformity."

But yes, providing good examples to start with would go a long way. As we
talked about building up a more complete standard library of device type
definitions, Ella pointed out that many manufacturers make NMEA definitions
available with their own names for the components, and that using these in
the "standard" definitions could be a start.