OceanDataTools/openrvdas #327: Allow serving all OpenRVDAS/Influx functions from default port

davidpablocohn, 2023-03-28:
We'd like to be able to tweak openrvdas_nginx.conf so that we only need one port (80 or 443) exposed to users, e.g.:

openrvdas/   - grafana
openrvdas/console - console
openrvdas/influx - influx

Everything plays well with this except Django, which currently requires changing all the rules in the urls.py file and all the links in the index.html, etc files. Would be nice to recode Django stuff so that we can tell it where it's going to get served from, and it'll use that as its base.

(Right now if we can change the urls.py file to add 'console/' to all the rules, and they'll find their targets. But when we click a button like "Load new definition" or one of the loggers, it'll try to reach its target page *without* the 'console/' prefix.

webbpinner, 2023-03-30:
For context...
My default recommendation for installing OpenRVDAS/Grafana/InfluxDB on vessel's is to breakout Grafana and InfluxDB onto it's own hardware/VM.  I have encountered too many instances where InfluxDB consumes system resources to the point that it compromises OpenRVDAS performance.

With these installs I typically setup Grafana to port 80 and Influx to 8086 (InfluxDB default).

davidpablocohn, 2023-03-30:
Excellent - thanks! You've also reminded me that I need to tweak
install_influxdb.sh so it can run on a machine that *doesn't* have
OpenRVDAS loaded.

FrozenGeek, 2023-06-12:
What is available on the outside of the firewall isn't necessarily what's available on the inside of the firewall.
You can leave grana/influx alone (as far as Django is concerned) and proxy them to the outside world
in the nginx config (pretty much similar to what's done with the CDS).

FrozenGeek, 2023-06-21:
Fired up influxdb and grafana so I can take a look at a couple approaches to the problem.  I'll post trials/tribulations/triumphs when I have something worth posting.

gabesargeant, 2024-04-07:
Hey, 
Adding onto FrozenGreeks' comment. 
Similar to how you've setup the proxy to the cached data server, 
https://github.com/OceanDataTools/openrvdas/blob/master/utils/install_openrvdas.sh#L645
you can do the same for graphana if it's running on the same server as the nginx instance, if you're wanting to put everything behind a https proxy and avoid having to use the port number in the url as http://rapahan:3000 
location / {
        proxy_pass http://graphana:3000;
        proxy_redirect off;
        proxy_set_header Host $host;
        proxy_set_header X-Real-IP $remote_addr;
        proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
        proxy_set_header X-Forwarded-Proto $scheme;
    }
 Cheers.

davidpablocohn, 2024-04-09:
Thanks for this, Gabe!

I'm going to leave the feature request open while we figure out how to
integrate the option to do this into the installation script. Or until I
manage to fold it into a doc somewhere.

(And great getting to spend time brainstorming and working with you and the
gang. Just got home last night, and am starting to dig out on backlog.)