OceanDataTools/openrvdas #337: Allow ws access when remotely accessing openrvdas webUI via port forwarding

webbpinner, 2023-07-20:
Several vessels provide remote access to the OpenRVDAS via ssh-based port forwarding.  This works for the static portions of the OpenRVDAS webUI but because the port used for the ws connection is explicitly defined to a port number within the  settings.py file the ws socket traffic is not accessible when accessing via ssh port-forwarding.

Running on the assumption that the WS port is the same as the port used by nginx during install would it be possible to define the ws port as the same port specified in the URL.

TL;DR
Can the default vault of `WEBSOCKET_SERVER` be changed from:
```
var WEBSOCKET_SERVER = "ws://:80/cds-ws";
```
To:
```
var WEBSOCKET_SERVER = (document.location.protocol === 'https:' ? 'wss' : 'ws' ) + '//:' + document.location.port + "/cds-ws";
```

davidpablocohn, 2023-07-20:
This looks reasonable, but I can't figure out which file this is in - not
django_gui/settings.py?

webbpinner, 2023-07-20:
It's in:
https://github.com/OceanDataTools/openrvdas/blob/master/django_gui/templates/django_gui/index.html
but that file just refers back to what's set in settings.py

davidpablocohn, 2023-07-20:
Got it. I'm not sure the text replacement you suggest would work, as even
though it ends up being used in JS code, it's handed to the JS code via
python, and first interpreted as a string in Python (in settings.py). And I
don't think Python would know what to do with document.location.port.

webbpinner, 2023-07-20:
I was curious about that.  My initial thought was that the string would be not be interpreted until it was in the browser which I assumed would be fine as just a JS string passed to the template via the Django code.  I might be wrong in that assumption.

LMG-ET, 2024-03-28:
We've solved that one.
We tweaked ${OPENRVDASROOT}/display/js/widgets/settings.js

var WEBSOCKET_PORT = window.location.port || 443;

That file may or may not have been generated by something further upstream (I'd have to run through the installation script and whatnot to give you a definitive answer), but ....

Standard disclaimer:  It works for me !!

davidpablocohn, 2024-04-09:
Ah, I think you can get away with this because the display widget port is
defined in JS code, but what Webb wants to change is specified in
django_gui/settings.py[.dist]. And Python can't handle the js variable.

I think the way to get this done would be to tweak the (static) js to
include a separate js config that's stashed somewhere canonical.

davidpablocohn, 2025-05-05:
Duplicate with #446

PR #447 (2025-05-05): Resolves #446 - Dynamically build cds-ws URL to account for accessing OpenRVDAS via port forwarding
This also resolves #337, right?
closed 2025-05-05 by davidpablocohn