OceanDataTools/openrvdas #347: Remove redundant UDP code from in NetworkWriter, finish/fix broken TCP code, and rename to TCPWriter

veggiemike, 2023-10-18:
Title says it all.  I haven't done this yet, so figured I'd make it an "issue" to discuss.  There's a ton of redundant functionality in NetworkWriter that I just finished cleaning up and thoroughly testing in UDPWriter.  Are you opposed to me removing all the UDP code from NetworkWriter?  Then all that's left is non-functional TCP code, which will be a lot easier to finish and maintain with the UDP removed.  Of course then the name is silly, so perhaps we rename it TCPWriter?

I'm itching to do it.

davidpablocohn, 2023-10-19:
I think that makes sense. I don't think anyone is using NetworkWriter. But rather than just disappearing it, I think the gentler transition ("Your mission, should you choose to accept it...") would be to replace the old NetworkWriter with a shell that says "This class is deprecated - please use TCPWriter or UDPWriter instead"

If you're up for that, I'd be very grateful.

I've accumulated a bunch of other small changes*, too, so once this is done, I think it makes sense to push out another versioned "master".

*The one I'm most excited about is supporting "--database sqlite" in the logger manager, so the entire thing can be run Django-free. Kevin Pedigo has a pure js/sqlite-based GUI that's still a little rudimentary but looks like an excellent template for creating a simpler way to drive the system.

veggiemike, 2023-10-20:
That sounds perfect to me!  I'll let you know if I run into any roadblocks!

davidpablocohn, 2023-12-02:
Merged to dev, master

closed 2023-12-02 by davidpablocohn