OceanDataTools/openrvdas #361: Install Script for Rocky Linux (9.3)

bitos2002, 2023-12-27:
Hello all !

I would like to ask you if someone has make the modifications on install_openrvdas.sh, for install openrvdas on Rocky Linux ?

Thanks !!!

davidpablocohn, 2023-12-27:
Hi Oscar - not yet, but it looks like a pretty simple fix. As I understand, the Rocky distributions correspond ("bug for bug" as they say) to the RHEL distros. I'm testing out a simple fix to install_openrvdas.sh (install_influxdb.sh) that should allow them to recognize Rocky, and if it works, I'll check in a change to incorporate that.

davidpablocohn, 2023-12-27:
Resolved with merge  f24d63c..39c3213  master -> master

closed 2023-12-27 by davidpablocohn

bitos2002, 2023-12-28:
Firstly , Pablo thanks for your help.
After a few minor problems with the installation i can say the script works correctly for Rocky Linux 9.3 installation.
I would like to comment one problem on python code.
I want to start the logger with the following command:
/opt/openrvdas/logger/utils/simulate_data.py --config test/NBP1406/simulate_NBP1406.yaml
There are some problem with the parse module.
---------
[rvdas@openrvdas openrvdas]$ logger/utils/simulate_data.py --config test/NBP1406/simulate_NBP1406.yaml
Traceback (most recent call last):
  File "/opt/openrvdas/logger/utils/simulate_data.py", line 89, in <module>
    import parse
ModuleNotFoundError: No module named 'parse'
---------
I have read that parse module is deprecated in python 3.9.
Do you have some modification for this issue ?
Thanks !!!

davidpablocohn, 2023-12-28:
Hi Oscar - I'm guessing that the problem you're running into is that, if
you're running simulate_data.py from the command line, you may not be
running it from within the virtual environment that the OpenRVDAS
installation set up. For compartmentalization, the installation script
creates a virtual environment and installs all the necessary packages
inside that. To use it, you can either activate the environment for your
current shell ('source venv/bin/activate'), or you can simply call the
python that is defined inside the venv ("venv/bin/python
logger/utils/simulate_data.py --config test/NBP1406/simulate_NBP1406.yaml")
).

Can you let me know if this solves your problem?

Also: to clarify - I believe it's the parser module that's deprecated, not
parse. The latter is what's used in OpenRVDAS.