OceanDataTools/openrvdas #367: Easier data retrieval from CDS

LMG-ET, 2024-03-28:
Currently if an external script (perhaps running on another machine) wants to grab a quick value from the CDS (i.e., What's the current air temperature) it involves a conversation: subscribe, ready, etc on a websocket.   While this is fine for data you expect to be monitoring for a while (like, say, a display), it's less useful for a shell script.

What might be a useful enhancement is the ability to query the most recent data value for a variable, like (e.g., 
https://openvdas.mydomain/CDS/variable_name) and get a string back (data_value)
 
Possible approach:
Since nginx can serve http and websockets from the same location block, on connection the CDS could check if it's receiving a normal HTTP request or a websocket connection, either returning a quick reply to a GET or proceeding as it does now for websocket traffic.

Alternative approach:
Since small and modular is often better than a monolithic swiss army knife for embeds like ROV's, that may not be the way to go.  A separate process (which you could optionally turn on with supervisord) that just gets proxied via nginx would work, so you'd do something like https://openrvdas.mydomain/latest_data/variable_name, you'd define a location block "latest_data/" in the nginx config, then nginx would proxy the request to a separate process,  That process could query the cached data server in the normal way.  Still the same conversation, but the complexity gets hidden from the end-user.

Alternate Alternate approach:
An nginx module to handle this functionality.  It seems like it might be both serious overkill and not in keeping with the python feng-shui of OpenRVDAS, but I don't think we'd find an approach with less latency.

What I'm specifically thinking about for this is getting data into elog.  USAP's current common configuration for elog automatically fills in things like latitude, longitude, depth, sea-surface temperature, etc when you create a new log entry.  Using a python script to talk to the CDS on the elog host would require installing the script, the correct modules (the websocket library, for example, is not part of the python standard library), so much better to have integral support for this kind of data query.

davidpablocohn, 2024-03-28:
Yesterday (or was it the day before?), while we were talking about reviving
a RESTful API for the control UI, we discussed whether it also made sense
to have something RESTful for the CDS. I don't think it would be all that
hard.

But I'm going to snooze this in my queue until I get back from a bit of
west Tasmania exploration next week.

LMG-ET, 2024-03-28:
> WARNING: This email originated from outside of USAP’s email system. Do not click links or open attachments without verifying with the sender and confirming the link or document is safe.

I’ve already done a crude implementation of option 2 here (just for testing), but if it didn’t add much to the size (both in-memory and on disk), I’d vote for option 1.

Option 3 would be fun, but it’s just silly.

 

--Kevin

yoLewshi, 2024-10-03:
@davidpablocohn I added something similar to this in the django rest api views on the contrib repo: https://github.com/OceanDataTools/openrvdas_contrib/blob/master/django_gui/api_views.py#L808 

Currently you need to be authenticated but it could be adapted to something simpler perhaps

webbpinner, 2025-05-06:
Does the LMG use InfluxDB/Grafana for dashboards?  If so then you might be able mimic what Sealog does:
https://www.oceandatatools.org/sealog-docs/server_influxdb_setup/

A more invasive solution would be to replace elog with Sealog which can probably be configured to output data in elog format.

More information on Sealog:
https://www.oceandatatools.org/sealog-docs/
https://www.linkedin.com/pulse/logging-deep-how-sealog-brings-structure-ocean-science-oceancensus-p2hee