OceanDataTools/openrvdas #376: Feature request: new parameter to tell CDS to bundle records

davidpablocohn, 2024-04-28:
Parameter should tell CachedDataServer to only return records every N seconds, and to return them as a bundle.

Requested by @ellapiet.

ellapiet, 2024-04-29:
@davidpablocohn I had a play with adding in that single line of asyncio.sleep() in the cached data reader. Looks like it isn't going to give us the functionality we were hoping for as records are still returned singularly rather than in a list

davidpablocohn, 2024-04-29:
Hmmmm - okay, thanks for poking at that. I'll have a look this week and see
if I'm able to kludge things in a not-too-kludgey way.

davidpablocohn, 2024-05-05:
@ellapiet - is there a case where you'd like to have different time window bundles for different variables (e.g. bundle heading every 10 seconds, but bundle temps every 30 seconds)? Or would it be adequate to set a single window for all the variables, e.g. "Give me all the DASRecords that I've subscribed to that have arrived in the last N seconds"?

If the latter, I *think* I can do a clean and easy implementation inside CachedDataReader.

davidpablocohn, 2024-05-05:
I think this change in branch issue_376 does what's needed:
https://github.com/OceanDataTools/openrvdas/blob/issue_376/logger/readers/cached_data_reader.py

One further improvement would be to base the bundle window on DASRecord
timestamps vs system time. Do you have feelings on which way that should go?

closed 2024-05-17 by davidpablocohn

ellapiet, 2024-05-19:
Sorry @davidpablocohn I'm out at sea right now and missed this one. Changes look great!