OceanDataTools/openrvdas #403: Websockets version 14 causing breaking changes in cached data server

ellapiet, 2024-11-15:
**Describe the bug**
cached_data_server.py line 967

`self.websocket_server = websockets.serve(
                ws_handler=self._serve_websocket_data,
                host='', port=self.port, loop=self.event_loop)`

ws_handler is having issues being parsed by websockets 14.

Our quick solution was to rollback to websockets 13 and add it to our requirements.txt file

davidpablocohn, 2024-11-15:
Right - I have this recollection that we already banged up against this
somewhere, and thought I (or someone else) had fixed it, but maybe it was
just in a dream? I can't find any trace. Problem is that WS14 doesn't want
the event loop argument, so we need to do some sort of conditional call.
The elegant thing to do would be to check version numbers, but that feels
fraught keeping compatibility between distributions. So quick and dirty
would be to have a try/except block here:

        try:
            self.websocket_server = websockets.serve(
                ws_handler=self._serve_websocket_data,
                host='', port=self.port, loop=self.event_loop)
        except [What's the exception thrown?]:
            self.websocket_server = websockets.serve(
                ws_handler=self._serve_websocket_data,
                host='', port=self.port)

Would you be willing to try throwing that in, or letting me know what
specific exception is getting thrown?

Many thanks,
 -p

davidpablocohn, 2024-11-28:
Hi Ella - did you have the chance to try the above patch to confirm that it
fixes the breakage you got with WS14?

angus-cummings, 2024-11-29:
Hi Pablo, 
The exception we're seeing is TypeError

```
Exception in thread Thread-2 (_run_websocket_server):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/openrvdas/server/cached_data_server.py", line 965, in _run_websocket_server
    self.websocket_server = websockets.serve(ws_handler=self._serve_websocket_data,
TypeError: serve.__init__() missing 1 required positional argument: 'handler'
```
We should be able to get around this particular complaint by just removing the keyword argument for (ws_)handler, they changed name from 13->14 but it's a positional argument so as long as its passed first nameless is fine. 

I'm still having issues getting it attached to the asyncio loop though, I'll keep investigating.

angus-cummings, 2024-11-29:
I'm not able to push this up and not confident it's ready for PR either, but the changes to cached_data_server.py attached (as .txt) seem to be working for me right now. I need to put it away for the weekend, just gifting it up in case you have interest before I get back to it. 

[cached_data_server.txt](https://github.com/user-attachments/files/17955027/cached_data_server.txt)

davidpablocohn, 2024-11-29:
Thank you! I'll give it a try and see if it works for me. I've replicated
the fault locally by using Python 3.13, and will test against that and the
good-old-fashioned Python 3.8-11.

davidpablocohn, 2024-11-30:
I had to make one tweak to your code to get it to work on legacy (3.8)
Python: add an (unused) loop parameter to _serve_websocket_data(): at line
997:

"""Top-level coroutine for running CachedDataServer."""
async def _serve_websocket_data(self, websocket, unused_loop_arg=None):
    # Legacy websocket code passes in event loop as third argument; we
don't need it,
    # but include so code works both pre and post WS14.

Other than that, it seem solid.

davidpablocohn, 2024-12-04:
Hi Angus - checking in whether you've had a moment (okay, probably takes a lot more than "a moment") to try this out on your setup?

angus-cummings, 2024-12-06:
Sorry Pablo, I was out walking for the last few days. I'll test it today.

angus-cummings, 2024-12-06:
They addition of the unused keyword argument doesn't break anything for us, but I have been struggling to understand why it's necessary. I even made a python3.8 venv to test it in, and I didn't run in to any issues... but there's a non-zero chance I've broken my test environment and need  a new one anyway.

davidpablocohn, 2024-12-06:
Thanks! And agreed - the extra arg requirement is a bit of a mystery to me,
too, but on the legacy setup I've got running as http://demo.openrvdas.org/
seems to barf unless I've got it there.

Would you like to file a pull request to dev, or shall I put the change in?

(Also: no apologies needed for being out - especially out walking! I know
we're all doing this as volunteers in our spare time.)

angus-cummings, 2024-12-07:
Perfectly happy for you to manage integrating the change, if that is ok with you Pablo!

davidpablocohn, 2024-12-07:
Will do!

davidpablocohn, 2024-12-08:
Merged `issue_403` into `dev`

commit 0c1e4284 (2024-12-08): Resolves #403. Websockets14 changes calling convention.
closed 2024-12-08 by davidpablocohn