OceanDataTools/openrvdas #405: Stopping logger_manager from supervisorctl causing logger shutdown failures

tking5, 2024-11-27:
**Describe the bug**

When stopping logger_manager via supervisorctl while running in no-console mode, loggers would correctly shut down but then a variable subset of loggers immediately restart. Conflicting execution pathways from logger_supervisor & logger_manager seem to be the cause. 
1. Logger_runner.py receives the interrupt from supervisorctl and raises a KeyboardInterrupt exception
2. Logger_manager catches then passes exception and calls logger_supervisor.quit() 
3. Logger_supervisor.quit() acquires a logger_map_lock, changes quit_flag to True, and correctly begins shutdown of all loggers
4. Simultaenously, logger_manager._update_configs() executes on update_configs thread loop, calling supervisor.update_configs(logger_configs) and waits for lock release from step 3
5. When lock is released, update_configs call from step 4 continues and executes critical section attempting to restart loggers as if they are new
6. Shutdown process completes, leaving a variable amount of loggers that were restarted in step 5 still running after logger_manager shutdown

Our startup sequence: `command=/opt/openrvdas/venv/bin/python server/logger_manager.py --database django --data_server_websocket :8766 -v -V --no-console --config ./local/coriolix/cruise.yaml`

Our shutdown command `supervisorctl stop logger_manager`

**Expected behavior**
Loggers should shutdown and not restart when update_configs thread resumes processing if quit flag was changed to True.

**Suggested fix**
logger_supervisor.py line 172

Our workaround was to wrap the critical section of logger_supervisor’s update_configs function with a check against self.quit_flag:

```
with self.logger_map_lock:
            if not self.quit_flag:
                    …
```

davidpablocohn, 2024-11-27:
Thanks for this, Tristan!

My fix at the moment is a slight variant of yours at line 172:

        with self.logger_map_lock:
            # If we're in the process of quitting, go home - a different thread is
            # already shutting things down.
            if self.quit_flag:
                return
           ...

Currently eyeballing whether a cleaner fix might be to simply put the "self.quit_flag = True" line at line 82 inside the lock that follows it. I don't see any harm in doing both, but would love it if you would sanity check my logic.

Thoughts?

tking5, 2024-12-02:
Both changes make sense to me! Without the early return the second thread would still proceed in error from logger_managers call to update_configs, unless I am missing something there, seems like both changes are required to prevent the behavior?

Thanks for addressing this for us so quickly!

davidpablocohn, 2024-12-02:
Merged fix to dev and master.

commit d47c3eca (2024-11-28): Resolves #405
closed 2024-12-02 by davidpablocohn