OceanDataTools/openrvdas #413: Script to "pickle" all parameters and definitions of current cruise

davidpablocohn, 2025-01-18:
Talking with R2R folks, they'd like to be able to capture a snapshot of what parameters, device definitions, configurations, etc are running during a cruise. This could be done with a script, either manually invoked, or somehow automatically invoked when something is changed, that captures the cruise definition file, all of the device definitions it's calling in, and the branch/version/timestamp of the OpenRVDAS code being run.

Script would package/pickle/tar the information into an easily storable file that could be archived with the data produced on a cruise.

Bonus functionality would be for it to (periodically?) do a diff and save a new timestamped copy if any of the relevant files have changed.

davidpablocohn, 2025-01-18:
> I kinda already do this for some boats. The logger configuration and all
> device files are archived every time a change is made.   The archives are
> timestamped and included as part of the OpenVDM cruise dataset. -WSent from

Excellent! Is that a script you can easily share? I'm going to try to put
some time into poking into .git files to see if there's an easy way to
snapshot the branch/date/etc of the repo, too.

webbpinner, 2026-10-23:
I'd suggest renaming this config_snapshot as not all openrvdas users use the word "cruise".  many use "expedition" and a couple use "voyage".  Same suggestion for "expanded_cruise" --> "expanded_config"

closed 2026-10-23 by davidpablocohn-claude-bot