OceanDataTools/openrvdas #415: override logging format of websocket package to use OpenRVDAS format

webbpinner, 2025-01-22:
log format used by cached_data_server.py differs from standard OpenRVDAS log format.
[image]

Request that format be consistent with other OpenRVDAS log messages.

davidpablocohn, 2025-01-24:
This is turning out to be a more subtle bug than I expected. The cached_data_server does have a standard logging format defined, but something in its includes is pre-empting and scuttling it. Still investigating.

davidpablocohn, 2025-01-25:
Okay - fixed in the branch. But it looks like there are *two* "standard" logging formats:

```
2025-01-24 17:20:57,760Z 20 INFO cached_data_server.py:1102 Starting CachedDataServer
2025-01-24 17:20:57,761Z 20 INFO cached_data_server.py:960 Starting WebSocketServer on port 8766
2025-01-24 17:20:57,767Z 20 INFO server.py:707 server listening on 0.0.0.0:8766
2025-01-24 17:20:57,767Z 20 INFO server.py:707 server listening on [::]:8766
```
and
```
2025-01-24 17:19:18,877 cached_data_server.py:1103 Starting CachedDataServer
2025-01-24 17:19:18,878 cached_data_server.py:961 Starting WebSocketServer on port 8766
2025-01-24 17:19:18,884 server.py:707 server listening on [::]:8766
2025-01-24 17:19:18,884 server.py:707 server listening on 0.0.0.0:8766
```
If I move us toward one or the other, which do you think is the right one?

webbpinner, 2025-01-25:
I personally like the one with the log level (the top one).

My rational...
This issue was first raised by SOI who are starting to analyze logs using 3rd-party tools.  Having the log level helps to easily classify the nature of the message (INFO, WARNING, ERROR, CRITICAL, DEBUG).

That said... what does the "20" column represent?

davidpablocohn, 2025-01-25:
It's the numeric log level :

'%(asctime)-15sZ',
'%(levelno)s',
'%(levelname)s',
'%(filename)s:%(lineno)d',
'%(message)s',

That format is defined in logger/utils/stderr_logging.py.

webbpinner, 2025-02-02:
I vote this move forward using the current format definition as defined in logger/utils/stderr_logging.py.  Users can override if needed.

davidpablocohn, 2025-02-10:
Resolved in dev

PR #419 (2025-02-10): Issue 415
Resolves #415
closed 2025-02-10 by davidpablocohn