OceanDataTools/openrvdas #422: Allow (parameterized) reusable logger definitions in cruise definition file

davidpablocohn, 2025-02-23:
Much the way devices and device types are reusable, the overhead of creating and maintaining a cruise definition file could be simplified if it was possible to define a few parameterized logger types and reuse them. 

It would operate along the lines of jinja, running a preprocessor over the initial file (perhaps automatically when the templatized file is loaded?) to expand it. We discussed some syntax conventions that seem like they would make this scale better than raw Jinja.

[image]

webbpinner, 2025-02-23:
My thoughts...
Combining config definitions into the loggers section makes sense.

Field name patterns for devices makes sense... I personally would prefer the default pattern to be %device%_%field%

Still on the fence with templates... I find the yaml native variable expansion to be a real improvement over what I've done in the past.  That said... I guess one way to implement would be:

Here's what goes in logger_config.yaml:
```
---
configs:
  cwnc-net:
    template: serial_to_net
      kwargs:
        prefix: %prefix%
        baudrate: %baudrate%
        serial_port: %serial_port%
```

Here's the template... written in it's own file(s) that are included with the logger_config via 'includes' OR written inline within the logger_config.yaml within a `templates` section:
```
---
templates:
  serial_to_net:
    readers:
    - class: SerialReader
      kwargs:
        baudrate: %baudrate%
        port: %serial_port%
    transforms:
    - class: TimestampTransform
    - class: PrefixTransform
      kwargs:
        prefix: %prefix%
    writers:
    - class: UDPWriter
      kwargs:
        port: 6224
        destination: 255.255.255.255
```

So the complete logger_config.yaml would look something like:
```
---
########################################
loggers:
  PCOD:
    configs:
    - PCOD-off: {}
    - PCOD-net:
       template: serial_to_net
          kwargs:
             prefix: PCOD
             baudrate: 4800
             serial_port: /tmp/tty_PCOD
  cwnc:
    configs:
    - cwnc-off: {}
    - cwnc-net:
       template: serial_to_net
          kwargs:
             prefix: cwnc
             baudrate: 19200
             serial_port: /tmp/tty_cwnc

########################################
templates:
  serial_to_net:
    readers:
    - class: SerialReader
      kwargs:
        baudrate: %baudrate%
        port: %serial_port%
    transforms:
    - class: TimestampTransform
    - class: PrefixTransform
      kwargs:
        prefix: %prefix%
    writers:
    - class: UDPWriter
      kwargs:
        port: 6224
        destination: 255.255.255.255
```

davidpablocohn, 2025-02-23:
Thanks! Yeah, that's pretty close to what Ella proposed - I've got a couple
of (my version of) her sample files I was going to add to the feature
request as soon as I cleaned up my typos.

davidpablocohn, 2025-02-27:
For including:
```
# Anything here is expected to be a YAML dict that gets merged/appended at the top level
# with this dict. If there are duplicate keys then:
#   - if value is scalar, it gets overwritten
#   - if value is list, it gets appended
#   - if value is dict, it gets merged
includes:
    /opt/openrvdas/investigator/configs/logger_templates/*.yaml
```

A couple of possibilities for templating, riffing off both Webb and Ella's ideas, depending on how much templatting we'd like to support. In theory, we could support both:
```
variables:
  source_ip: 224.0.36.0
  eol: '\n'
  udp_loggers:
    - pco2
    - knud
    - s330

modes:
      '%udp_loggers%': '%udp_loggers%-off'
      met: met-off
      pco2: pco2-off
    log+db:
      '%udp_loggers%': '%udp_loggers%-on'
      met: met-on
      pco2: pco2-on

loggers:
  pco2:
    logger_template: standard_udp
    variables:
      port: 54321
  ...
```

Alternatively, could use variable expansion all the way down:
```
variables:
  source_ip: 224.0.36.0
  eol: '\n'

  # or,alternatively, each could be a dict with its own defs:
  udp_loggers:
    pco2:
      logger_template: standard_udp
      logger_port: 54321
    knud:
      logger_template: standard_udp
      logger_port: 54322
    s330:
      logger_template: standard_udp
      logger_port: 54323

modes:
      '%udp_loggers%': '%udp_loggers%-off'
      met: met-off
      pco2: pco2-off
    log+db:
      '%udp_loggers%': '%udp_loggers%-on'
      met: met-on
      pco2: pco2-on

loggers:
  # Gets expanded using variables
  '%udp_loggers%':
    logger_template: '%logger_template%'
    port: '%logger_port%'
```

webbpinner, 2025-02-27:
I'd like to see more complete examples.  I'm unclear how the examples in that last post will scale for some of the more complicated installations like Falkor, OceanXplorer and Revelle.

I'm also not feeling the `modes` section syntax.  What about including the modes in the logger config as a scalar or array:
```
  cwnc:
    configs:
    - cwnc-off:
        modes: off
    - cwnc-net:
        modes: [net, net+influx]
```

the modes section could then be built automatically from the logger configs

webbpinner, 2025-02-27:
ok... this is gonna be a long one but I needed to piece together a complete example based on the NBP config.  Here's where I got:

Here's the sample logger config:
```
---
net_writer: &net_writer_defaults
  port: 6224
  destination: 255.255.255.255

cwnc: &cwnc_defaults
  prefix: cwnc
  baudrate: 19200
  serial_port: /tmp/tty_cwnc

pcod: &pcod_defaults
  prefix: pcod
  baudrate: 4800
  serial_port: /tmp/tty_PCOD

########################################
loggers:

  pcod:
    configs:
    - pcod-off: { modes: off }
    - pcod-net:
        modes: [net, net+influx]
        template:
          serial_to_net:
            kwargs:
              <<: *pcod_defaults
    - pcod-net+file:
        modes: [net+file, net+file+influx]
        template:
          serial_to_net+file:
            kwargs:
              <<: *pcod_defaults
              filepath: /var/tmp/log/pcod/raw/<<cruise>>_pcod

  cwnc:
    configs:
    - cwnc-off: { modes: off }
    - cwnc-net:
        modes: [net, net+influx]
        template:
          serial_to_net:
            kwargs:
              <<: *cwnc_defaults
    - cwnc-net+file:
        modes: [net+file, net+file+influx]
        template:
          serial_to_net+file:
            kwargs:
              <<: *cwnc_defaults
              filepath: /var/tmp/log/cwnc/raw/<<cruise>>_cwnc

  influx:
    configs:
    - influx-off: { modes: [ off, net, net+file] }
    - influx-on: { modes: [net+influx, net+file+influx] }
      readers:
        - class: UDPReader
          <<: *net_writer_defaults
      writers:
        - class: InfluxDBWriter

########################################
templates:
  serial_to_net: &serial_to_net
    readers:
[... 211 lines cut ...]
  net+influx:
    pcod: pcod-net
    cwnc: cwnc-net
    influx: influx-on
  net+file:
    pcod: pcod-net+file
    cwnc: cwnc-net+file
    influx: influx-off
  net+file+influx:
    pcod: pcod-net+file
    cwnc: cwnc-net+file
    influx: influx-on
```

This may even help with that WHOI request of no modes...

I'm not trying to derail any previous discussions but I think adding looping capability WITHIN the logger config deserves more thought/discussion.

This approach offers a lot of power while preserving valid YAML syntax and keeping the flexibility that most ships require.  It also limits the code changes to just the template implementation.

As for looping, I would vote to do that upstream of the logger config like how the jinja implementation was done and even how my logger building script works.

This approach eliminates everything but the `loggers` section.  The defaults objects and templates object can be moved to different files and added back in as includes. This makes building the loggers section via a pre-processing script easy... in fact maybe the jinja code should be brought to bear for that step.

Thoughts?

davidpablocohn, 2025-02-27:
Innnteresting - thanks!

I'm not at all wedded to the looping thing - it just seemed like a handy shortcut.

I like the double-angle-brackets notation; obviates the need to have so many things quoted.

Not as keen on having each config say which modes it belongs to - that feels less intuitive and more error prone to me. But let me think on it a bit more?

Also, one minor note: in your configs: section, you're using a list of dicts, where I think you can just use dicts? (Unless I'm missing something)
```
  pcod:
    configs:
    - pcod-off: { modes: off }
    - pcod-net:
        modes: [net, net+influx]
        template:
          serial_to_net:
            kwargs:
              <<: *pcod_defaults
    - pcod-net+file:
        modes: [net+file, net+file+influx]
        template:
          serial_to_net+file:
            kwargs:
              <<: *pcod_defaults
              filepath: /var/tmp/log/pcod/raw/<<cruise>>_pcod
```
could be simplified as
```
  pcod:
    configs:
      pcod-off: { modes: off }
      pcod-net:
        modes: [net, net+influx]
        template:
          serial_to_net:
            kwargs:
              <<: *pcod_defaults
      pcod-net+file:
        modes: [net+file, net+file+influx]
        template:
          serial_to_net+file:
            kwargs:
              <<: *pcod_defaults
              filepath: /var/tmp/log/pcod/raw/<<cruise>>_pcod
```
right?

webbpinner, 2025-02-28:
Moving from list to dict is absolutely the way to go as this would also prevent configs with duplicate names.

I agree with your comments on the modes.  I was making a compromise between defining modes within the config vs it's own section (current model).  In the end the current modes sections gets "built" from the configs so I'm guessing/hoping the logger_manager would first look for a modes section in the config and, if not found, would try to build it from the modes scalar/list within the config... and fail to load if neither is available.

yoLewshi, 2025-03-03:
I think assuming the installation directory with an option to override that through the key makes sense and should be flexible enough. It might also allow some standard "chunks" to be included in the repo (where one sample might "include" another file). Full paths is clearer but less flexible.

davidpablocohn, 2025-03-07:
So here's a sample templated cruise definition. I've got the code implemented in branch issue_422 to transparently read and expand this into a full cruise definition. There's a lot more that can be done, such as making the true_winds_template use variables, and I've omitted the "modes" section, since it's just as long and rambling as before. But if you don't count the standard logger templates (which in theory don't need to be modified), it takes about 1500 lines down to about 160 lines (260 if you count the as-yet-unmunged modes section).

I also include serial_logger_template.yaml below

```
###########################################################
###########################################################
# YAML cruise definition file for OpenRVDAS.
#
# Created by:
#   command:  <<command_line>>
#   time:     <<date_time>> UTC
#   user:     <<user>>
#
########################################
cruise:
  id: <<cruise>>
  start: '<<cruise_start>>'
  end: '<<cruise_end>>'

###################
includes:
- local/logger_templates/serial_logger_template.yaml
- local/logger_templates/udp_logger_template.yaml
- local/logger_templates/parse_data_logger_template.yaml
- local/logger_templates/true_winds_logger_template.yaml
- local/logger_templates/snapshot_logger_template.yaml

########################################
# Global variables - can be overridden by individual loggers
variables:
  cruise: NBP1406
  cruise_start: 2019-07-01
  cruise_end: 2019-12-31
  
  udp_destination: 255.255.255.255
  raw_udp_port: 6224
  parsed_udp_port: 6225
  
  file_root: /var/tmp/log
  parse_definition_path: test/NBP1406/devices/nbp_devices.yaml

  data_server: localhost:8766
  influx_bucket_name: openrvdas
  
  command_line: local/usap/create_template_cruise.py test/NBP1406/NBP1406_port_defs.yaml
  date_time: Fri Mar  7 21:28:45 2025
  user: pablo

########################################
loggers:
  PCOD:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_PCOD
      baud_rate: 4800
  cwnc:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_cwnc
      baud_rate: 19200
  gp02:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_gp02
      baud_rate: 4800
  gyr1:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_gyr1
[... 126 lines cut ...]
      net+file:
        readers:
        - class: SerialReader
          kwargs:
            baudrate: <<baud_rate>>
            port: <<serial_port>>
        transforms:
        - class: TimestampTransform
        writers:
        - class: LogfileWriter
          kwargs:
            filebase: <<file_root>>/<<logger>>/raw/<<cruise>>_<<logger>>
        - class: ComposedWriter
          kwargs:
            transforms:
            - class: PrefixTransform
              kwargs:
                prefix: <<logger>>
            writers:
              - class: UDPWriter
                kwargs:
                  port: <<raw_udp_port>>
                  destination: <<udp_destination>>

```

davidpablocohn, 2025-03-09:
Now implemented in #issue_422 branch; resulting cruise config (in test/NBP1406/NBP1406_template_cruise.yaml) is about 270 lines vs ~1510, about 18% of the size. I still want to work on streamlining the first pass on logger templates, in local/logger_templates (I remember we talked about moving them somewhere else?).

davidpablocohn, 2025-03-10:
Here's my draft update to the documentation on cruise definition files, that I *think* captures the new (backward compatible) proposed change. Would love to hear what folks think:

============
### Templated Cruise Definition Files

The traditional way of specifying a cruise definition file is...verbose. The sample `NBP1406_cruise` is over 1500 lines, and and difficult to maintain or modify. To date, different institutions have coped with this by generating these files mechanically, either from customized scripts or databases. We have recently implemented a templatized format for specifying cruise definitions that promises to drastically reduce this complexity. Templatized cruise definitions may be loaded the same way as traditional cruise definitions - OpenRVDAS will recognize and expand them internally.

#### Logger Templates

The crucial element of a templatized cruise definition is a logger template. Many loggers in a typical installation will be almost identical, varying only by a few fields, such as ports, or prefixes to be attached.

The [serial_logger_template](https://github.com/OceanDataTools/openrvdas/blob/master/local/logger_templates/serial_logger_template.yaml) encodes a typical set of logger configurations for a serial-port-based instrument:
```buildoutcfg
###################
logger_templates:

  #################
  serial_logger_template:
    configs:
      'off': {}
      
      # Read, timestamp, prefix, and send off to net via UDP
      net:
        readers:
        - class: SerialReader
          kwargs:
            baudrate: <<baud_rate>>
            port: <<serial_port>>
        transforms:
        - class: TimestampTransform
        - class: PrefixTransform
          kwargs:
            prefix: <<logger>>
        writers:
        - class: UDPWriter
          kwargs:
            port: <<raw_udp_port>>
            destination: <<udp_destination>>
            
      # Write both to network and save to file
      net+file:
        readers:
        - class: SerialReader
          kwargs:
            baudrate: <<baud_rate>>
            port: <<serial_port>>
        transforms:
        - class: TimestampTransform
        writers:
        - class: LogfileWriter
          kwargs:
            filebase: <<file_root>>/<<logger>>/raw/<<cruise>>_<<logger>>
        - class: ComposedWriter
          kwargs:
            transforms:
            - class: PrefixTransform
              kwargs:
                prefix: <<logger>>
            writers:
              - class: UDPWriter
                kwargs:
                  port: <<raw_udp_port>>
                  destination: <<udp_destination>>

```

Note the names enclosed in `<<double_brackets>>`; these will be filled in by variables on a logger-by-logger basis in the cruise definition file:

```buildoutcfg
###########################################################
# Global variables - can be overridden by individual loggers
variables:
  cruise: NBP1406
  raw_udp_port: 6224
  udp_destination: 255.255.255.255  
  file_root: /var/tmp/log
 
###########################################################
loggers:
  PCOD:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_PCOD
      baud_rate: 4800
  cwnc:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_cwnc
      baud_rate: 19200
```
The logger templates may either be included directly in the cruise definition file under the `logger_templates:` key, as above, or included by reference from an `includes:` section:
```buildoutcfg
###########################################################
includes:
  - local/logger_templates/serial_logger_template.yaml
  - local/logger_templates/parse_data_logger_template.yaml
  - local/logger_templates/true_winds_logger_template.yaml
  - local/logger_templates/snapshot_logger_template.yaml
```
Note that filepaths specified in the `includes` block may include wildcards such as `*`, and the files may themselves have an `includes` section. All logger templates are merged into the top-level dict, with later keys overwriting earlier ones.

#### Modes

The `modes` section - at present, remains unchanged and relatively verbose, with one important difference. If no `modes` key is found, the system will try to infer one. It will create a single default mode such that every logger is running the first configuration defined for it (in the case of `serial_logger_template`, this would be `off`).

This simplication allows institutions who have only one operating mode (and rely on the system to simply 'come up logging') to define loggers with only one configuration and simply omit creating a 'modes' section.

If there _are_ other configurations, they will still be accessible via the GUI or command line interface. The only difference is that they will be have to activated on a per-logger basis, rather than all at once using cruise mode selection.

webbpinner, 2025-03-11:
For installations that have a lot of one-off config profiles...  Is it still possible to define logger configs inline and without the use of a template?

i.e.
```
###########################################################
loggers:
  PCOD:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_PCOD
      baud_rate: 4800
  cwnc:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_cwnc
      baud_rate: 4800
  geo_fencing:
    configs:
      off: {}
      on:
      readers: {class: UDPReader, kwargs: {port: 7221}}
      transforms:
        - class: ParseTransform
          kwargs:
            strip_unprintable: true
            definition_path: /opt/openrvdas/local/devices.yaml
        - class: GeofenceTransform
          module: logger.transforms.geofence_transform
          kwargs:
            latitude_field_name: seapath380MRU_Latitude
            longitude_field_name: seapath380MRU_Longitude
            boundary_dir_name: /home/rvdas/enabled_eez_gml_files
            leaving_boundary_message: set_active_mode underway
            entering_boundary_message: set_active_mode eez
      writers:
        - class: LoggerManagerWriter
          module: logger.writers.logger_manager_writer
          kwargs:
            database: django
            allowed_prefixes:
              - 'set_active_mode '
              - 'sleep '
```

davidpablocohn, 2025-03-11:
Yes - that works, too. I should update the docs to say that, and perhaps to
describe the sequence of expansion: first, any templates are expanded into
inline definitions, then any inline definitions are propagated to the
cruise definition's "configs" section.

davidpablocohn, 2025-03-11:
One could also argue that for conciseness and readability, even without
variables, it might make sense to have the geofence logger in its own file,
nominally as a template. So perhaps 'logger_template' is the wrong name -
perhaps it should be 'logger_type'? And a logger could be defined in one of
three ways:

1. With a configs declaration (and configs to follow in the configs
section):

  PCOD:
    configs:
    - "off"
    - net
    - net_write

2. With a full configs definition:

  PCOD:
    configs:
      "off": {}
      net:
        ...
      net_write :
        ...

3. Or by reference to a pre-defined logger_type:

  PCOD:
    logger_type: serial_logger
    variables: # optional, if that logger_type includes templatized
variables

webbpinner, 2025-03-12:
This is where I'm struggling a little bit.
I definitely like the template concept but I don't know how I feel about this level of hiding configs behind included files.  The real power in OpenRVDAS is it's modularity; the ability to define loggers, build configs for those loggers and map configs to modes... none of that is being diminished/replaced but I feel this hides too much of what's actually happening.

You are right about the 'logger_template' name.  At what point are they no longer template and just a part of the logger_config.yaml file stored in a different file?

When I did the super long post earlier I focused on templates being templates: skeleton yaml objects for building configs (not loggers).

```
pcod: &pcod_defaults
  prefix: pcod
  baudrate: 19200
  serial_port: /tmp/tty_pcod

pcod:
    configs:
      pcod-off: {}
      pcod-net:
        template:
          serial_to_net:
            kwargs:
              <<: *pcod_defaults
      pcod-net+file:
        template:
          serial_to_net+file:
            kwargs:
              <<: *pcod_defaults
              filepath: /var/tmp/log/pcod/raw/<<cruise>>_pcod
```

While not NEARLY as concise as:
```
  pcod:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_pcod
      baud_rate: 4800
```

...I like the readability of seeing the individual configs, the flexibility to override variables where/when needed and the ability to add cruise-specific configs for loggers when requested by the science party (<-- this happens).

This also (in my opinion) makes it clear that a template is a template for building a config.

I really like what you've already done which simplifies my initial proposal.

```
---
###########################################################
# Global variables - can be overridden by individual loggers
variables:
  cruise: NBP1406
  file_root: /var/tmp/log
  raw_udp_port: 6224
  udp_destination: 255.255.255.255

########################################
loggers:
  pcod:
    configs:
      pcod-off: {}
      pcod-net:
        template:
          serial_to_net:
            kwargs:
              baudrate: 4800
      pcod-net+file:
        template:
          serial_to_net+file:
            kwargs:
              baudrate: 4800

  cwnc:
    configs:
      cwnc-off: {}
      cwnc-net:
        template:
          serial_to_net:
            kwargs:
              baudrate: 19200
      cwnc-net+file:
        template:
          serial_to_net+file:
            kwargs:
              baudrate: 19200

  influx:
    configs:
      influx-off: {}
      influx-on:
        readers:
          - class: UDPReader
            port: <<raw_udp_port>>
        writers:
          - class: InfluxDBWriter

########################################
templates:
  serial_to_net:
    readers:
    - class: SerialReader
      kwargs:
        baudrate: <<baudrate>>
        port: /tmp/tty_<<logger>>
    transforms:
    - class: TimestampTransform
    - class: PrefixTransform
      kwargs:
        prefix: <<logger>>
    writers:
    - class: UDPWriter
      kwargs:
        port: <<raw_udp_port>>
        destination: <<udp_destination>>

  serial_to_net+file:
    readers:
    - class: SerialReader
      kwargs:
        baudrate: <<baudrate>>
        port: /tmp/tty_<<logger>>
    transforms:
    - class: TimestampTransform
    writers:
    - class: LogfileWriter
      kwargs:
        filebase: <<file_root>>/<<logger>>/raw/<<cruise>>_<<logger>>
    - class: ComposedWriter
      kwargs:
        transforms:
        - class: PrefixTransform
          kwargs:
            prefix: <<logger>>
        writers:
        - class: UDPWriter
          kwargs:
            port: <<raw_udp_port>>
            destination: <<udp_destination>>
```

Would be great to hear more from our Southern Hemisphere users.

davidpablocohn, 2025-03-14:
Ahyup. The new code maintains complete backward compatibility, so you can
inline things if/where you want. I made one more change yesterday/today:
now instead of requiring modes to be specified as dicts, you can specify
them as lists of configs. The code will look up the corresponding loggers
and make sure that there's a config for every logger, then expand it into
the "normal" mode definition. So this works:
```

###########################################################
modes:
  'off':  # without a 'default_mode' key, first defined mode will be default
    [PCOD-off, cwnc-off, gp02-off, gyr1-off, adcp-off, eng1-off,
svp1-off, twnc-off, mbdp-off, knud-off, grv1-off, mwx1-off, pco2-off,
pguv-off, s330-off, tsg1-off, rtmp-off, hdas-off, tsg2-off, seap-off,
parse_data-off, true_winds_port-off, true_winds_stbd-off,
snapshot-off]

  no_write:
    [PCOD-net, cwnc-net, gp02-net, gyr1-net, adcp-net, eng1-net,
svp1-net, twnc-net, mbdp-net, knud-net, grv1-net, mwx1-net, pco2-net,
pguv-net, s330-net, tsg1-net, rtmp-net, hdas-net, tsg2-net, seap-net,
parse_data-on, true_winds_port-on, true_winds_stbd-on, snapshot-on]

  write:
    [PCOD-net+file, cwnc-net+file, gp02-net+file, gyr1-net+file,
adcp-net+file, eng1-net+file, svp1-net+file, twnc-net+file,
mbdp-net+file, knud-net+file, grv1-net+file, mwx1-net+file,
pco2-net+file, pguv-net+file, s330-net+file, tsg1-net+file,
rtmp-net+file, hdas-net+file, tsg2-net+file, seap-net+file,
parse_data-on, true_winds_port-on, true_winds_stbd-on, snapshot-on]

  no_write+influx:
    [PCOD-net, cwnc-net, gp02-net, gyr1-net, adcp-net, eng1-net,
svp1-net, twnc-net, mbdp-net, knud-net, grv1-net, mwx1-net, pco2-net,
pguv-net, s330-net, tsg1-net, rtmp-net, hdas-net, tsg2-net, seap-net,
parse_data-on+influx, true_winds_port-on+influx,
true_winds_stbd-on+influx, snapshot-on+influx]

  write+influx:
    [PCOD-net+file, cwnc-net+file, gp02-net+file, gyr1-net+file,
adcp-net+file, eng1-net+file, svp1-net+file, twnc-net+file,
mbdp-net+file, knud-net+file, grv1-net+file, mwx1-net+file,
pco2-net+file, pguv-net+file, s330-net+file, tsg1-net+file,
rtmp-net+file, hdas-net+file, tsg2-net+file, seap-net+file,
parse_data-on+influx, true_winds_port-on+influx,
true_winds_stbd-on+influx, snapshot-on+influx]

```

This also makes use of the 'missing default_mode' feature that - while not
recommended, you can leave that out and the loader will select the first
defined mode as the default mode. Makes it really easy for simple cruises:
if you omit the modes definition, loader will synthesize a 'default' mode
using the first defined config for each logger. Then because it's the
'first' (and only) mode, it will select that as the default mode to be run
when the system starts up.

I've written (most of?) it up in the draft doc update at
https://davidpablocohn.github.io/openrvdas-docs/cruise_definition_files/

But yeah - I'd love to hear what our antipodal colleagues think of this
approach.

Thanks!

ellapiet, 2025-03-14:
Sorry, currently at sea. Have just caught up reading and I like what I currently see in [https://davidpablocohn.github.io/openrvdas-docs/cruise_definition_files/](url). I see @webbpinner  has mentioned that he wants too see each individual config (e.g. 'off', net, net+file, etc), but for me this increases the repetition so much in the config file that it makes the templates potentially not worth it. I have my whole team coming on and potentially needing to change these config files. So I really just want them to see what they need in one place (i.e.not defaults of each logger at the top) and once (i.e. not all the different modes with the only change being an added database writer). So all in that to say I like the current proposal and don't think it should change too much from that. @davidpablocohn I noticed in your draft that there is no mention of the modes section of the template

davidpablocohn, 2025-03-14:
Thanks, and apologies for prodding you at sea! There's no urgency on this;
mostly trying to get it squared away, but I don't need to commit/pull it
anytime soon.

Re: modes - in the documentation I kept all the discussion of modes up in
the "traditional" section, just adding a paragraph about the option for
compact representation and about inferring modes/default if the section is
missing. I'll see if I can clean up the docs to make the whole thing a bit
clearer.

But I'm going to let it sit for a little while I deal with non-OpenRVDAS
things and check back in in a couple of weeks after I've had time to mull.

Thanks again, all!

webbpinner, 2025-03-21:
Just curious...

For the ships I manage, I've configured the data management system to package and archive the logger_config.yaml, device files, devices.yaml, and the template file my script uses to build the logger_config.yaml file.  A new archive is built any time a change is made to any of the files.  The archive filenames includes the cruiseID and timestamp and the archives for a given cruise are included in the final cruise data package delivered to the science party and archives.

Is there a way (maybe via API call) to export the logger_config.yaml file in it's complete traditional format from the new template/include system?

davidpablocohn, 2025-03-21:
Ahyup:

logger/utils/read_config.py  test/NBP1406_cruise.yaml > expanded_yaml.py

webbpinner, 2025-03-22:
After studying this config file expansion architecture (elegant as always), It looks like it should be possible to add a "config_template" expansion between:
```
    result = expand_logger_templates(result)
    result = expand_config_templates(result)  # <-- new expansion
    result = expand_logger_definitions(result)
```
This would provide not only the config templates that would benefit the more irregular deployments but would also allow for even shorter logger_templates (because they could potentially leverage config_template expansion).

I'm not asking for you to work anymore on this... you've already done plenty.  I'm mainly just asking whether or not I'm understanding your code correctly.

davidpablocohn, 2025-03-22:
No worries, and thanks! If there's a way to improve things, I'm all in. To
make sure I'm following you: this would be in
read_config.py:expand_cruise_definition, right? But I've lost the thread of
exactly what expand_config_templates() would provide - it's like the
logger_templates, but for individual configs, right?

webbpinner, 2025-03-24:
Correct.
So in addition to this syntax:
```
########################################
loggers:
  PCOD:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_PCOD
      baud_rate: 4800

###################
logger_templates:

  #################
  serial_logger_template:
    configs:
      'off': {}
      
      # Read, timestamp, prefix, and send off to net via UDP
      net:
        readers:
        - class: SerialReader
          kwargs:
            baudrate: <<baud_rate>>
            port: <<serial_port>>
        transforms:
        - class: TimestampTransform
        - class: PrefixTransform
          kwargs:
            prefix: <<logger>>
        writers:
        - class: UDPWriter
          kwargs:
            port: <<raw_udp_port>>
            destination: <<udp_destination>>
            
      # Write both to network and save to file
      net+file:
        readers:
        - class: SerialReader
          kwargs:
            baudrate: <<baud_rate>>
            port: <<serial_port>>
        transforms:
        - class: TimestampTransform
        writers:
        - class: LogfileWriter
          kwargs:
            filebase: <<file_root>>/<<logger>>/raw/<<cruise>>_<<logger>>
        - class: ComposedWriter
          kwargs:
            transforms:
            - class: PrefixTransform
              kwargs:
                prefix: <<logger>>
            writers:
              - class: UDPWriter
                kwargs:
                  port: <<raw_udp_port>>
                  destination: <<udp_destination>>
```

This syntax would also work:
```
########################################
loggers:
  PCOD:
    configs:
      pcod-off: {}
      pcod-net:
        config_template: serial_to_net_template
          variables:
            serial_port: /tmp/tty_PCOD
            baud_rate: 4800
      pcod-net+file:
        config_template: serial_to_net_and_file_template
          variables:
            serial_port: /tmp/tty_PCOD
            baud_rate: 4800

###################
config_templates:

  #################
  serial_to_net_template: &serial_to_net_template_base
    readers:
      - class: SerialReader
        kwargs:
          baudrate: <<baud_rate>>
          port: <<serial_port>>
    transforms:
      - class: TimestampTransform
    writers:
      - class: ComposedWriter
        transforms:
          - class: PrefixTransform
            kwargs:
              prefix: <<logger>> 
        writers:
          - class: UDPWriter
            kwargs:
              port: <<raw_udp_port>>
              destination: <<udp_destination>>

  serial_to_net_and_file_template:
  <<: *serial_to_net_template_base
    writers:
      - class: LogfileWriter
        kwargs:
          filebase: <<file_root>>/<<logger>>/raw/<<cruise>>_<<logger>>
      - class: ComposedWriter
        kwargs:
          transforms:
            - class: PrefixTransform
              kwargs:
                prefix: <<logger>>
            writers:
              - class: UDPWriter
                kwargs:
                  port: <<raw_udp_port>>
                  destination: <<udp_destination>>
```

and optionally the config_templates could be used to build logger_templates i.e.:
```
###################
config_templates:

  #################
  serial_to_net_template: &serial_to_net_template_base
    readers:
      - class: SerialReader
        kwargs:
          baudrate: <<baud_rate>>
          port: <<serial_port>>
    transforms:
      - class: TimestampTransform
    writers:
      - class: ComposedWriter
        transforms:
          - class: PrefixTransform
            kwargs:
              prefix: <<logger>> 
        writers:
          - class: UDPWriter
            kwargs:
              port: <<raw_udp_port>>
              destination: <<udp_destination>>

  serial_to_net_and_file_template:
  <<: *serial_to_net_template_base
    writers:
      - class: LogfileWriter
        kwargs:
          filebase: <<file_root>>/<<logger>>/raw/<<cruise>>_<<logger>>
      - class: ComposedWriter
        kwargs:
          transforms:
            - class: PrefixTransform
              kwargs:
                prefix: <<logger>>
            writers:
              - class: UDPWriter
                kwargs:
                  port: <<raw_udp_port>>
                  destination: <<udp_destination>>

###################
logger_templates:

  #################
  serial_logger_template:
    configs:
      'off': {}
      
      # Read, timestamp, prefix, and send off to net via UDP
      net:
        config_template: serial_to_net_template
          variables:
            serial_port: <<serial_port>>
            baud_rate: <<baud_rate>>
            
      # Write both to network and save to file
      net+file:
        config_template: serial_to_net_and_file_template
          variables:
            serial_port: <<serial_port>>
            baud_rate: <<baud_rate>>
```

davidpablocohn, 2025-03-24:
Got it. Hmmm - I like the idea. Now trying to figure out whether it makes
sense to just generalize things so that there are just "templates". It
wouldn't be that hard to allow nested templates so that a template for a
logger could include templates for configs. Upside: avoids creating yet
another type of entity to keep track of. Downside: also easier to create
malformed loggers (by specifying a logger template where you meant to put a
config template)

webbpinner, 2025-04-12:
Just to recap these changes...

In addition to what previously worked... this now also works:
```
###########################################################
loggers:
  mwx1:
    logger_template: serial_logger_template
    variables:
      serial_port: /tmp/tty_mwx1
      baud_rate: 9600
  s330:
    configs:
      'off': {}
      net:
        config_template: serial_net_config_template
        variables:
          baud_rate: 9600
          serial_port: /tmp/tty_s330
      net+file:
        config_template: serial_net_file_config_template
        variables:
          baud_rate: 9600
          serial_port: /tmp/tty_s330
```
where `serial_logger_template` looks like:
```
###################
logger_templates:
  #################
  serial_logger_template:
    configs:
      'off': {}
      net:
        config_template: serial_net_config_template
        variables:
          baud_rate: <<baud_rate>>
          serial_port: <<serial_port>>
          raw_udp_port: <<raw_udp_port>>
          udp_destination: <<udp_destination>>
      net+file:
        config_template: serial_net_file_config_template
        variables:
          baud_rate: <<baud_rate>>
          serial_port: <<serial_port>>
          raw_udp_port: <<raw_udp_port>>
          udp_destination: <<udp_destination>>
```

and `serial_reader_config_template` looks like:
```
###################
config_templates:
  #################
  serial_net_config_template: &serial_net_config_base
    readers:
    - class: SerialReader
      kwargs:
        baudrate: <<baud_rate>>
        port: <<serial_port>>
    transforms:
    - class: TimestampTransform
    writers:
    - class: ComposedWriter
      kwargs:
        transforms:
        - class: PrefixTransform
          kwargs:
            prefix: <<logger>>
        writers:
        - class: UDPWriter
          kwargs:
            port: <<raw_udp_port>>
            destination: <<udp_destination>>

  serial_net_file_config_template:
    <<: *serial_net_config_base
    writers:
    - class: LogfileWriter
      kwargs:
        filebase: <<file_root>>/<<logger>>/raw/<<cruise>>_<<logger>>
    - class: ComposedWriter
      kwargs:
        transforms:
        - class: PrefixTransform
          kwargs:
            prefix: <<logger>>
        writers:
        - class: UDPWriter
          kwargs:
            port: <<raw_udp_port>>
            destination: <<udp_destination>>
```

PR #424 (2025-03-20): Issue 422
Resolves #422 ; also a bunch of smaller fixes that should have been merged separately:
- move test NBP data to its own subdir
- allow implicit conversion of DASRecord to JSON if transform or writer can take a str but not a DASRecord
- cleaned up whitespace
closed 2025-04-15 by davidpablocohn