OceanDataTools/openrvdas #423: Minimal cruise configuration should be an empty file

davidpablocohn, 2025-02-23:
Currently, the logger manager barfs if a cruise configuration file is missing cruise dates, a cruise name, or other fields. In theory, it should be able to do just fine with any/all of those fields missing.

davidpablocohn, 2025-02-26:
This is turning out to be more difficult than I'd expected. Much of the API assumes modes (specifically, default modes).

webbpinner, 2025-02-26:
I wasn't in Hobart so I have to ask...  why?

Dropping the requirements for cruiseID, and/or start/stop dates are reasonable requests... but modes/default mode?  What is the driver for this request?

davidpablocohn, 2025-02-26:
It's a combination of a couple of things. One was that you shouldn't need cruise id, dates, etc. and the other was folks like WHOI saying that they just want everything running when OpenRVDAS is running, so modes are an unnecessary level of abstraction. The first of these is pretty easy; I think I've already got that bit of code fixed. The latter *seemed* like it should be easy (if modes aren't defined, can, e.g. say that when system starts up, all loggers go to their first defined configuration, or say behavior is undefined).

But it turns out that the assumption that modes are defined is *everywhere* in the code and the API (not clever design, Pablo). So I'll probably just fix things so that it's fine without the cruise variables. (The way forward with modes would be for the system to generate a 'modes' section automatically if one isn't present, but place where modes start to get expected is at the API level, so it *might* be necessary to fix it in every API implementation, which feels...challenging.)

webbpinner, 2025-02-26:
How many lines of yaml code does removing `modes` / `default_mode` save?  I'm guessing about:
```
 modes:
  write+influx:
    PCOD: PCOD-net+file
    cwnc: cwnc-net+file
    gp02: gp02-net+file
    gyr1: gyr1-net+file
    adcp: adcp-net+file
    eng1: eng1-net+file
    svp1: svp1-net+file
    twnc: twnc-net+file
    mbdp: mbdp-net+file
    knud: knud-net+file
    grv1: grv1-net+file
    mwx1: mwx1-net+file
    pco2: pco2-net+file
    pguv: pguv-net+file
    s330: s330-net+file
    tsg1: tsg1-net+file
    rtmp: rtmp-net+file
    hdas: hdas-net+file
    tsg2: tsg2-net+file
    seap: seap-net+file
    net_reader: net_reader-on+influx
    true_wind: true_wind-on+influx
    snapshot: snapshot-on+influx

########################################
default_mode: 'write+influx'
```

Doesn't seem worth it.  Also... WHOI may decide in the future they'll actually want modes for dealing with things like transiting through foreign EEZ zones.

my $0.02

davidpablocohn, 2025-02-26:
It's a good point, but I think you've just given me an idea about an easy
way forward: this could be pretty easily combined with the logger template
feature. The implementation of templating would be loading the raw file and
running a preprocessor over it to load 'include' files and fill out
templates. It would be pretty simple for that same preprocessor to create
and auto-populate 'modes' and 'cruise' sections if they weren't present.

webbpinner, 2025-02-26:
So something like:

```
modes:
  default:
    {%logger%: %logger_default_config% for %logger% in %loggers%}

default_mode: 'default'
```

davidpablocohn, 2025-03-20:
Merged into #422

closed 2025-03-20 by davidpablocohn