OceanDataTools/openrvdas #440: RecordParser field_patterns dict should only match on matching key

davidpablocohn, 2025-04-15:
When field_patterns is a dict, RecordParser should check the record's data_id against the keys of the dict and only try to match patterns under that key. Currently it tries to match all patterns, regardless of key.

```
              # Wind mast
              wind_mast_port:
              - "$WIMWV,{PortWindDegreesRel:g},R,{PortWindSpeedRel:g},{PortDataValid:w},{Mode:w}*{CheckSum:x}"
              wind_mast_stbd:
              - "$WIMWV,{StbdWindDegreesRel:g},R,{StbdWindSpeedRel:g},{StbdDataValid:w},{Mode:w}*{CheckSum:x}"

```

davidpablocohn, 2025-04-15:
It's unclear what behavior we want if the record is missing a data_id (try all patterns?).

closed 2025-04-15 by davidpablocohn

webbpinner, 2025-04-15:
If a record is missing a data_id won't it automatically fail because the record won't match the `record_format: '{data_id:w} {timestamp:ti} {field_string}'` kwarg?

davidpablocohn, 2025-04-15:
Righto - good point. I've also just added an additional warning
specifically for when the data_id doesn't match any key, so users will be
able to distinguish that from cases where it matches a key, but no patterns
under the key.

commit 8b044d34 (2025-04-15): Resolves #440 (#441)
commit 37188fb1 (2025-04-15): Resolves #440