OceanDataTools/openrvdas #446: Dynamcally build cds-ws URL to account for accessing OpenRVDAS via port forwarding

webbpinner, 2025-04-30:
Accessing OpenRVDAS via port forwarding doesn't work because the CDS websocket URL is statically defined.

This could be made dynamic and adjust based on the django gui URL with the following patch:
```
--- a/django_gui/static/django_gui/websocket.js
+++ b/django_gui/static/django_gui/websocket.js
@@ -28,6 +28,8 @@ var websocket_server = WEBSOCKET_SERVER;
 if (websocket_server.indexOf('//:') > 0) {
   websocket_server = websocket_server.replace('//:', '//' + location.hostname + ':');
 }
+const url = new URL(websocket_server);
+websocket_server = url.protocol + '//' + window.location.hostname + (window.location.port ? ':' + window.location.port : '') + url.pathname;
```

davidpablocohn, 2025-05-01:
Thanks for this - I think it's a resurfacing of #337, right? Which I got distracted from and dropped on the floor.

I'll give your solution a try on a couple of systems and try to fold it in before Monday.

davidpablocohn, 2025-05-05:
Merged to dev

commit 0ca7a5ca (2025-05-04): Resolves #446.
PR #447 (2025-05-05): Resolves #446 - Dynamically build cds-ws URL to account for accessing OpenRVDAS via port forwarding
This also resolves #337, right?
commit 3fd236ba (2025-05-05): Resolves #446 - Dynamically build cds-ws URL to account for accessing OpenRVDAS via port forwarding (#447)

* Resolves #446.

* Resolves #446.
closed 2025-05-05 by davidpablocohn