OceanDataTools/openrvdas #451: split_interval argument in FileWriter causes a type error

BenAtWHOI, 2025-05-27:
**Description**  
Adding the `split_interval` argument to a `FileWriter` causes a `TypeError` when invoked.

**To Reproduce**  
Steps to reproduce the behavior:
1. Create a configuration file
2. Add a `FileWriter` to the `writers` section and add a `split_interval` argument, e.g. `1H`

```
(...other configuration)
writers:
- class: FileWriter
  kwargs:
    filename: /path/to/output/file
    split_by_time: True
    split_interval: 1H
```

3. Invoke the configuration file, e.g. `python logger/listener/listen.py --config path/to/config.yaml`
4. View the error output:

```
Traceback (most recent call last):
  File "/opt/openrvdas/logger/listener/listen.py", line 885, in <module>
    listener.run()
  File "/opt/openrvdas/logger/listener/listener.py", line 112, in run
    raise e
  File "/opt/openrvdas/logger/listener/listener.py", line 99, in run
    self.writer.write(record)
  File "/opt/openrvdas/logger/writers/composed_writer.py", line 97, in write
    self.writers[0].write(record)
  File "/opt/openrvdas/logger/writers/file_writer.py", line 285, in write
    new_file_suffix = self._get_file_suffix()
  File "/opt/openrvdas/logger/writers/file_writer.py", line 224, in _get_file_suffix
    timedelta(seconds=self.split_interval_in_seconds))
TypeError: unsupported type for timedelta seconds component: str
```

**Expected behavior**  
The function `_get_split_interval_in_seconds` should properly parse the `split_interval_in_seconds` value and cause the timedelta calculation to not return an error. Currently, in the constructor on lines 122-123, the constructor parses the integer and the `H` or `M` from the value of `split_interval`, but then passes in the original unparsed string rather than the tuple with an integer and string:

```
self.split_interval = (int(split_interval[:-1]), split_interval[-1])
self.split_interval_in_seconds = self._get_split_interval_in_seconds(split_interval)
```

This is the correct way to do it:

```
self.split_interval = (int(split_interval[:-1]), split_interval[-1])
self.split_interval_in_seconds = self._get_split_interval_in_seconds(self.split_interval)
```

**More context**  
As far as I know, this was always a problem - we just added the `split_interval` argument to `FileWriter` for the first time recently. This can be reliably reproduced, any `split_interval` argument will cause the error due to how the code currently handles it.

OpenRVDAS is being run using the master branch on an Ubuntu 22.04 server.

davidpablocohn, 2025-05-27:
Whoops! That's embarrassing - many thanks for the catch, and for the
detailed fix. I'll try to get this patched this evening.

BenAtWHOI, 2025-05-27:
No problem! :)

commit 33799a41 (2025-05-28): Resolves #451 and expands unittest to verify. (#452)
commit 051d3019 (2025-05-28): Resolve #451 (#453)

* Issue 416 (#418)

* WIP - initial import, still need to test

* WIP - slight refactor to add filtering

* added requested functionality from issue #416

* code cleanup, expanded documentation

* updated inline documentation

* Flake8 related changes

* Get CDS to actually use standard logging format (#419)

* Issue 422 (#424)

* #423

* #422 - first pass at recursive include-and-merge

* #422 - updated unittest for read_config

* #422 - allow wildcards in includes

* Improve error logging, add temp diagnostics for #422

* #422 - use project base dir as default

* #422 al [... cut]